Pulse-Eight/libcec #366: Samsung TV not reacting on CEC command - Ubuntu Mate 16.04 Raspberry Pi 2

Paluschki, 2017-08-01:
**Setup:**
- Samsung TV UE40H6470
- Raspberry Pi 2 Model B
- connected via HDMI cable

Installed libcec by the instructions from [here](https://bugs.launchpad.net/ubuntu/+source/libcec/+bug/1655425):
> So a work-around that worked for me on Ubuntu-Mate 16.04: I purged the old versions of cec-utils and libcec3. Then I downloaded the .deb packages from the "Raspbian" repositories and manually installed those. Here are the packages I installed for Mate-16.04 (in this order)
1: libp8-platform2_2.1.0.1 ~ jessie_armhf.deb
2: libcec4_4.0.2.2 ~ jessie_armhf.deb
3: cec-utils_4.0.2.2 ~ jessie_armf.deb

**Problem:**
Samsung TV does not react on commands. In this case the TV does not turn off. Samsung TV does not find the Rapberry via AnyNet+. Interestingly the TV **was** able to find the Raspberry as I used the compiled version of libcec before (more information see below debug information).

Commandline:
`sudo echo "standby 0" | cec-client -s`

opening a connection to the CEC adapter...
DEBUG:   [             325]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [             326]	Open - vc_cec initialised
DEBUG:   [             326]	logical address changed to Free use (e)
NOTICE:  [             327]	connection opened
DEBUG:   [             327]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [             327]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [             327]	<< e0
DEBUG:   [             327]	processor thread started
DEBUG:   [             358]	>> POLL sent
DEBUG:   [             358]	TV (0): device status changed into 'present'
DEBUG:   [             358]	<< requesting vendor ID of 'TV' (0)
TRAFFIC: [             358]	<< e0:8c
DEBUG:   [            1449]	expected response not received (87: device vendor id)
TRAFFIC: [            1449]	<< e0:8c
DEBUG:   [            2511]	expected response not received (87: device vendor id)
NOTICE:  [            2511]	registering new CEC client - v4.0.2
DEBUG:   [            2511]	detecting logical address for type 'recording device'
DEBUG:   [            2511]	trying logical address 'Recorder 1'
DEBUG:   [            2511]	<< Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [            2511]	<< 11
TRAFFIC: [            2782]	<< 11
DEBUG:   [            3052]	>> POLL not sent
DEBUG:   [            3052]	using logical address 'Recorder 1'
DEBUG:   [            3052]	Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [            3052]	Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [            3052]	Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [            3052]	Recorder 1 (1): CEC version 1.4
DEBUG:   [            3052]	AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [            3053]	logical address changed to Recorder 1 (1)
DEBUG:   [            3053]	Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [            3053]	Recorder 1 (1): menu language set to 'eng'
DEBUG:   [            3054]	GetPhysicalAddress - physical address = 3000
DEBUG:   [            3054]	AutodetectPhysicalAddress - autodetected physical address '3000'
DEBUG:   [            3054]	Recorder 1 (1): physical address changed from ffff to 3000
DEBUG:   [            3054]	<< Recorder 1 (1) -> broadcast (F): physical address 3000
TRAFFIC: [            3054]	<< 1f:84:30:00:01
NOTICE:  [            3204]	CEC client registered: libCEC version = 4.0.2, client version = 4.0.2, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 3.0.0.0, git revision: libcec-4.0.2+8-8563411~dirty, compiled on Sun Apr  2 17:04:01 UTC 2017 by root@hostname: Name or service not known on Linux 4.4.0-57-generic (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
DEBUG:   [            3205]	<< Recorder 1 (1) -> TV (0): OSD name 'CECTester'
TRAFFIC: [            3205]	<< 10:47:43:45:43:54:65:73:74:65:72
DEBUG:   [            3505]	<< requesting power status of 'TV' (0)
TRAFFIC: [            3506]	<< 10:8f
DEBUG:   [            4567]	expected response not received (90: report power status)
TRAFFIC: [            4567]	<< 10:8f
DEBUG:   [            5628]	expected response not received (90: report power status)
DEBUG:   [            5629]	<< requesting vendor ID of 'TV' (0)
TRAFFIC: [            5629]	<< 10:8c
DEBUG:   [            6690]	expected response not received (87: device vendor id)
TRAFFIC: [            6690]	<< 10:8c
DEBUG:   [            7751]	expected response not received (87: device vendor id)
NOTICE:  [            7751]	<< putting 'TV' (0) in standby mode
TRAFFIC: [            7751]	<< 10:36
DEBUG:   [            7812]	unregistering all CEC clients
NOTICE:  [            7812]	unregistering client: libCEC version = 4.0.2, client version = 4.0.2, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 3.0.0.0, git revision: libcec-4.0.2+8-8563411~dirty, compiled on Sun Apr  2 17:04:01 UTC 2017 by root@hostname: Name or service not known on Linux 4.4.0-57-generic (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
DEBUG:   [            7812]	Recorder 1 (1): power status changed from 'on' to 'unknown'
DEBUG:   [            7813]	Recorder 1 (1): vendor = Unknown (000000)
DEBUG:   [            7813]	Recorder 1 (1): CEC version unknown
DEBUG:   [            7813]	Recorder 1 (1): osd name set to 'Recorder 1'
DEBUG:   [            7813]	Recorder 1 (1): device status changed into 'unknown'
DEBUG:   [            7813]	unregistering all CEC clients
DEBUG:   [            8330]	UnregisterLogicalAddress - releasing previous logical address
DEBUG:   [            8330]	logical address changed to Broadcast (f)
``

**Possible error source:**
At first I was able to compile libcec 4.0.2 but I made some mistakes during the compilation process. cec-client was missing the "rPi" support and I got the "autodetect failed" error. Then I manually deleted cec-client, libcec and platform cloned from github. I also deleted the created files in /usr/local. Could this be the cause of the error, because I did not successfully "make uninstall" libcec at first and so there are still parts of my compiled version interfering with the files from the packages? If yes, how do I properly remove compiled libcec from my system?

opdenkamp, 2017-08-28:
I don't see how this can be a problem in libCEC. Your TV acks a poll sent to it and doesn't reject any of the commands. Check whether CEC support (Anynet+) is enabled on your TV and check settings like "auto standby".

You might want to upgrade your git tree and rebuild, because you're using a version from a couple of months ago, but I don't see anything that would cause this in the log. And don't compile as user 'root' :)

closed 2017-08-28 by opdenkamp

Paluschki, 2017-09-04:
You were right, there was no problem with libcec.

Samsung AnyNet seemed to had a problem. I deactivated AnyNet, turned off the Samsung TV, unplugged it from grid, plugged it again and restarted AnyNet. Then it found the RaspberryPi via AnyNet and the commands do work now. Maybe this helps someone with a Samsung TV having the same issue.