Pulse-Eight/libcec #369: Windows: cec-tray crashes with kodi

ThomDietrich, 2017-08-02:
Hey,
CEC is configured and able to turn the TV on and off.
As soon as Kodi is started (or while it is running in the background) the cec-tray application closes itself without further warning. What's going on, which additional data do you need from me to tackle this issue?

Windows 10, Kodi 17.0

opdenkamp, 2017-08-28:
Because only one program can use libCEC, the tray application will automatically close. cec-tray's integration with Kodi is very basic (only r/c pass through will work) and Kodi has it's own integration, supporting more libCEC features.

If you want to use CEC on Windows while not using Kodi and want more control over what happens, then I suggest that you take a look at EventGhost, for which we provide a plugin. Just run libCEC's installer another time after installing EventGhost.

closed 2017-08-28 by opdenkamp

ThomDietrich, 2017-08-28:
Okay thanks for the tip. I'll have a look.

Regarding "the tray application will automatically close" - this is (as can be seen by this issue) not intuitive. The application should not close itself without user feedback. I'm most probably not the only one riddled by this behavior. You should consider changing this. Maybe an option would be to disable the application, indicated by another tray icon.

Side note: it's common to await a response before closing an issue. The goal should be to solve the "issue" (in this case the impression of a crash) before marking it as solved.

ThomDietrich, 2017-08-30:
I forgot to mention the worst thing about the "crash": The application doesn't restart after Kodi was closed. Yes I can disable the kodi addon, still this should not happen. I'd appreciate if you'd reopen the issue and reconsidered the situation at hand.

ThomDietrich, 2017-09-27:
@opdenkamp ping

kidforpm, 2021-01-20:
sorry to strat up this old thread,
Is there a way to restart the tray application as soon as Kodi exits?
The remapped keypresses I use in Windows don't work unless the server is running