Pulse-Eight/libcec #374: Autodetect fails for unprivileged user

JKierstead, 2017-08-11:
This issue is occurring using libcec 4.0.2 as built through Portage, as well as using the master branch as of commit 8adc786bac9234fc298c941dd442c3af3155a522. The system is a PC running Gentoo with the 4.8.17-hardened-r2 kernel.

When executing cec-client via sudo or through root, the application is able to autodetect my adapter, and works just fine:

```
~ $ sudo cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.2
no serial port given. trying autodetect: 
 path:     /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1
 com port: /dev/ttyACM0

opening a connection to the CEC adapter...
DEBUG:   [              19]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              20]	connection opened, clearing any previous input and waiting for active transmissions to end before starting
DEBUG:   [              25]	communication thread started
DEBUG:   [             136]	turning controlled mode on
NOTICE:  [             306]	connection opened
...
```

However, when I run as an unprivileged user, autodetect fails:
```
~ $ cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.2
no serial port given. trying autodetect: FAILED
```

Apart from a grsec entry about cec-client being executed, no information about the execution is logged.

My user is in the UUCP group associated with the TTY device. Specifying the device works for an unprivileged user:

```
~ $ cec-client /dev/ttyACM0 
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.2
opening a connection to the CEC adapter...
DEBUG:   [               0]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [               1]	connection opened, clearing any previous input and waiting for active transmissions to end before starting
DEBUG:   [               6]	communication thread started
DEBUG:   [             116]	turning controlled mode on
NOTICE:  [             287]	connection opened
...
```

opdenkamp, 2017-08-11:
The detection code uses udev https://github.com/Pulse-Eight/libcec/blob/master/src/libcec/adapter/Pulse-Eight/USBCECAdapterDetection.cpp#L351
It requires no special permissions other than that.

Your hardened/grsec system is probably blocking udev access, which could be caused by many things. I can only help by pointing to the code (see link above), but other than that, you'll have to find out yourself what's blocked where exactly when you're using a setup like this.

closed 2017-08-11 by opdenkamp

JKierstead, 2017-08-12:
Thanks for the tip, it was a grsec issue. A review of my kernel config showed I had accidentally enabled CONFIG_GRKERNSEC_SYSFS_RESTRICT, which blocked user access to everything under /sys.