Pulse-Eight/libcec #376: cec-client not working on Raspberry Pi

hamstu, 2017-08-20:
On a Raspberry Pi 1 Model B

```
$ lsb_release -a
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.1 (stretch)
Release:	9.1
Codename:	stretch
```

I followed [all the instructions](https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md) for building on my pi and after it complete, running `cec-client` I get:

```
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.1
no serial port given. trying autodetect: FAILED
```

I know the HDMI cable and TV support it because I was _just_ using it before wiping the SD card installing the most recent raspbian. (Previously it was an wheezy, I think.)

Interstingly, if I run `cec-client -i` I get this:

```
libCEC version: 4.0.1, compiled on Linux-3.8.13.30 ... , features: P8_USB, DRM, P8_detect, randr
```

Shouldn't `RPi` be in the features?

Any help would be much appreciated!

Sytanek, 2017-08-20:
I'm actually fighting the exact same thing today, first time using a Pi (Zero W) and trying to use libCEC.

libCEC version: 4.0.1, compiled on Linux-3.8.13.30 ... , features: P8_USB, DRM, P8_detect, randr

hamstu, 2017-08-21:
Hey @Sytanek! Glad someone else can relate, haha, and also because I have good news that I came back to report! 🎉  Hopefully this helps you with your Pi Zero. 

I went ahead and wiped my RPi's SD card, then I flashed onto it an older version of Raspbian; I decided on the last version of [Raspbian Jessie](http://ftp.jaist.ac.jp/pub/raspberrypi/raspbian/images/raspbian-2017-07-05/) I could find. Then I followed the build instructions as before. This time I encountered the exact issues described in #362, but following the recommended file patch in that thread I was able to complete the build, and now CEC/cec-client is working again!

```
pi@raspberrypi:~ $ cec-client -l
libCEC version: 4.0.2, git revision: libcec-4.0.2+30-8adc786~dirty, compiled on Sun 20 Aug 22:40:26 UTC 2017 by pi@raspberrypi on Linux 4.9.35+ (armv6l), features: P8_USB, DRM, P8_detect, randr, RPi
Found devices: 1

device:              1
com port:            RPI
vendor id:           2708
product id:          1001
firmware version:    1
type:                Raspberry Pi
```

Seems like something is off with latest version of Raspbian (stretch). Though my knowledge of all things compilers, etc. is too limited to know exactly what. Anyway, let me know if this helps you!

Sytanek, 2017-08-21:
Thanks for the update hamstu! Very interesting that it compiled as 4.02 this time, instead of 4.01.

Sytanek, 2017-08-21:
@hamstu I actually got it to work on latest. Needed to remove libcec4-dev which was evidently installed in full raspbian stretch? I didn't even have to recompile, now when I run it reports as 4.0.2 and is working as expected.

I found the issue by trying to install the .deb for 4.0.2 and it threw the error about libcec4-dev only being on version 4.0.1, so I removed it, and then ran my compiled bins, and they worked proper. This is the deb I used. https://www.deb-multimedia.org/pool/main/libc/libcec-dmo/libcec-dev_4.0.2-dmo1_armhf.deb

hamstu, 2017-08-21:
Ah, very interesting @Sytanek! Glad you got things sorted on your end too. 👍

opdenkamp, 2018-03-15:
If you're compiling from source and want to run it, then you'll have to tell the system to pick the compiled library instead of the one that's installed on the system. Use `LD_LIBRARY_PATH` to override the default library search path

closed 2018-03-15 by opdenkamp