Pulse-Eight/libcec #377: cec-client failing on Raspbian stretch

dbghb, 2017-08-21:
After updating to stretch (on a Pi 3) cec-client fails with
"cec-client: symbol lookup error: /usr/lib/arm-linux-gnueabihf/Utils6FormatEPKczlibcec.so.4: undefined symbol: _ZN11String"

This seems to be a recurring bug. 

I also compiled from source but the same problem shows up.

retikulumx, 2017-08-21:
Same for me. Looks like the same issue as in [#149](https://github.com/Pulse-Eight/libcec/issues/149)

retikulumx, 2017-08-24:
Ok, the issue is solved in current update (Debian Stretch Repository). Just update & upgrade

dbghb, 2017-08-24:
Verified.

closed 2017-08-24 by dbghb

fcanepa, 2017-12-02:
I'm currently running uptodated Debian Stretch on my rpi2, and I've the same error. How to fix rpi2 ?

retikulumx, 2017-12-02:
Just update libcec

fcanepa, 2017-12-02:
Ok, my apt.source.d was wrong... Now libcec is update to 4.0.2.3~stretch and it works!