Pulse-Eight/libcec #379: Is Windows 10 - ia32 supported?

kingio, 2017-09-06:
**O.S**: Windows 10
**Arch**: ia32

I wanted to use this library: https://www.npmjs.com/package/node-cec which attempts to use `cec-client`
I installed the libcec and i couldn't find `cec-client.exe` , so after a search i tried to use the ones from https://github.com/MediaBrowser/emby-theater-windows/tree/master/ThirdParty/cec
but says that non of them are compatible.
¿How can i use libcec?¿Where is `cec-client.exe` ?

ValorNaram, 2018-01-29:
Generally, you shouldn't use Windows operating systems for such things. Learn linux, because its a lot easier to use system. I think you want to turn Windows to a server, which isn't a good idea, because Windows itself is not a server software. Linux is capable of both Desktop and Server running

kingio, 2018-02-02:
@ValorNaram don't you think that if linux was an option i would be already using it?

mikevmil, 2018-02-15:
@ValorNaram Never heard of Windows Server?

ValorNaram, 2018-02-15:
@mikevmil Heard about Windows Server, but Windows is bad at all and Microsoft ignores existing standarts and creates their own. From the windows history I can see that Microsoft has stolen many things from another systems and implemented many linux feactures. And Windows is ever ten years older than linux, because Microsoft implements new feactures as last one. One could say feactures will be programmed under linux and in linux implemented and then 5 or 10 years later Microsoft implements the same feacture in their systems. So Windows is a downgrade compared to linux.

opdenkamp, 2018-03-15:
Sorry, I can't help with external tools. If it just requires cec-client.exe, download libCEC from our website, and enable 'cec debugging client' when installing: https://www.pulse-eight.com/Download/Get/51

closed 2018-03-15 by opdenkamp