Pulse-Eight/libcec #381: Raspberry 3 in 4k mode: Samsung TV not present

clohse, 2017-09-10:
I've tuned my /boot/config.txt to output 4k to my Samsung Smart TV (SW: 1501):
```
disable_overscan=1
framebuffer_width=3840 
framebuffer_height=2160
max_framebuffer_width=3840
max_framebuffer_height=2160
#hdmi_ignore_edid=0xa5000080
hdmi_pixel_freq_limit=400000000
hdmi_group=2
hdmi_mode=87
hdmi_timings=3840 1 48 32 80 2160 1 3 5 54 0 0 0 24 0 211190000 3
```

With this setup, cec-client (4.0.2) is not working:
> not sending command 'standby': destination device 'TV' marked as not present

Commenting out one of the parameters breaks the 4k setup. Commenting out all parameters brings me back the 2k setup with cec-client working fine.

I've tried some other options like hdmi_drive=2, hdmi_ignore_cec_init=1,  hdmi_ignore_cec=1, hdmi_force_hotplug=1... no luck either.

Raspberry 3 Model B:
```
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.1 (stretch)
Release:	9.1
Codename:	stretch
```
and `cec-client -i`:
```
libCEC version: 4.0.2, git revision: libcec-4.0.2+30-8adc786~dirty, compiled on Mon Aug 21 09:41:41 UTC 2017 by root@hostname: Name or service not known on Linux 4.4.0-92-generic (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
```

johnjore, 2017-10-30:
Me too!

Almost same issue with a Toshiba 4K TV, 49U5766DB and a RPi 1.

Config file with the relevant sections for either 1080p (CEC working) and 4K, (CEC not working)
```
# 1080p
#hdmi_group=1
#hdmi_mode=16

#4K
hdmi_group=2
hdmi_mode=87
hdmi_cvt 3840 2160 24
hdmi_pixel_freq_limit=400000000
max_framebuffer_width=3840
max_framebuffer_height=2160
```

I used apt-get install cec-utils on RPI Wheezy to install:
```
cec-client -i
libCEC version: 2.2.0, host: armv7l-unknown-linux-gnueabihf, features: 'P8 USB' 'P8 USB detect' 'RPi' 'EXYNOS', compiled on: Tue Nov  4 00:38:22 UTC 2014 by pbuilder@mgorven-pc.thefacebook.com on Linux 2.6.32 (armv7l)
```

I suspect this is not related to client-cec, but RPI firmware / custom HDMI mode 87 not enabling the CEC interface. Thoughts?

Klausoleum, 2018-02-19:
is there any news concerning the problem?

opdenkamp, 2018-11-02:
this is not a problem in libCEC, but the Pi, which doesn't fully support 4K, and that custom HDMI mode probably disables or breaks CEC indeed. we can't fix this in libCEC.

closed 2018-11-02 by opdenkamp