Pulse-Eight/libcec #384: Determine the HDMI port number currently used by the TV

smartguy1, 2017-09-22:
Is there a way to **detect the HDMI input** (port) being currently used by the TV? 
I am attempting to trigger a program when the user switches HDMI inputs. However, I cannot seem to find a command (high-level or raw Tx) that returns the HDMI port number being currently used. 
Has anyone attempted this?

trinitronx, 2018-01-16:
I'm definitely no expert at CEC, and have just started to pick this up... but I believe that HDMI-N ports correspond to the "Physical Address".  I found a good [doc for CEC here](https://linuxtv.org/downloads/presentations/media_summit_2016_san_diego/cec-wide_1.pdf)

In that pdf, there is an image showing how physical addresses are related to port & device chains.  In the simplest case with 1 device per HDMI port "chain", you'd end up with the HDMI-N port being equal to each device's physical address top-level number.

For example:

- `HDMI-1` = `1.0.0.0`
- `HDMI-2` = `2.0.0.0`
- HDMI-3 = `3.0.0.0`
- `HDMI-4` = `4.0.0.0`

It gets more complicated with more devices chained to each sub-device.  See the image on pdf **`page 3`** for a more complex example.  (`PA` means Physical Address in the pdf image, and is not separated by the dots)

opdenkamp, 2018-03-15:
Correct, thanks for answering :)

closed 2018-03-15 by opdenkamp

trinitronx, 2018-03-15:
@opdenkamp : Ok, glad to see I'm understanding it correctly!  If I could ask for a moment to understand some odd behavior I'm seeing with some code using [a CGO binding][1] to `libcec`.  I've posted details in #402

[1]: https://github.com/chbmuc/cec/blob/master/libcec.go

opdenkamp, 2018-03-15:
It's caused by that wrapper not providing a valid hdmi port configuration, which hits a bug in libCEC. Fix is on it's way

trinitronx, 2018-03-15:
@opdenkamp :  Thanks for quick reply!  Ok, so if I understand correctly:  This is indeed a bug, and once fixed I just need to rebuild against new `libcec`?

opdenkamp, 2018-03-16:
Correct. Providing a valid HDMI port config should fix the issue too indeed.