Pulse-Eight/libcec #389: old cec code not working with libcec 4

superuser789, 2017-09-29:
I am running stretch on raspberry pi 1. Only libcec4 and libcec4-dev is available in repo.

Simple code I found from [github][1] is based on old version of libcec. 

    // Build command:
    // g++-4.8 -std=gnu++0x -fPIC -g -Wall -march=armv6 -mfpu=vfp -mfloat-abi=hard -isystem /opt/vc/include/ -isystem /opt/vc/include/interface/vcos/pthreads/ -isystem /opt/vc/include/interface/vmcs_host/linux/ -I/usr/local/include -L /opt/vc/lib -lcec -lbcm_host -ldl cec-simplest.cpp -o cec-simplest 
    //#CXXFLAGS=-I/usr/local/include
    //#LINKFLAGS=-lcec -ldl
    #include <libcec/cec.h>
    
    // cecloader.h uses std::cout _without_ including iosfwd or iostream
    // Furthermore is uses cout and not std::cout
    #include <iostream>
    using std::cout;
    using std::endl;
    #include <libcec/cecloader.h>
    
    #include "bcm_host.h"
    //#LINKFLAGS=-lbcm_host
    
    #include <algorithm>  // for std::min
    
    // The main loop will just continue until a ctrl-C is received
    #include <signal.h>
    bool exit_now = false;
    void handle_signal(int signal)
    {
        exit_now = true;
    }
    
    
    //CEC::CBCecKeyPressType 
    int on_keypress(void* not_used, const CEC::cec_keypress msg)
    {
        std::string key;
        switch( msg.keycode )
        {
            case CEC::CEC_USER_CONTROL_CODE_SELECT: { key = "select"; break; }
            case CEC::CEC_USER_CONTROL_CODE_UP: { key = "up"; break; }
            case CEC::CEC_USER_CONTROL_CODE_DOWN: { key = "down"; break; }
            case CEC::CEC_USER_CONTROL_CODE_LEFT: { key = "left"; break; }
            case CEC::CEC_USER_CONTROL_CODE_RIGHT: { key = "right"; break; }
            default: break;
        };
    
        std::cout << "on_keypress: " << static_cast<int>(msg.keycode) << " " << key << std::endl;
        return 0;
    }
    
    
    int main(int argc, char* argv[])
    {
        // Install the ctrl-C signal handler
        if( SIG_ERR == signal(SIGINT, handle_signal) )
        {
            std::cerr << "Failed to install the SIGINT signal handler\n";
            return 1;
        }
    
        // Initialise the graphics pipeline for the raspberry pi. Yes, this is necessary.
        bcm_host_init();
    
        // Set up the CEC config and specify the keypress callback function
        CEC::ICECCallbacks        cec_callbacks;
        CEC::libcec_configuration cec_config;
        cec_config.Clear();
        cec_callbacks.Clear();
    
        const std::string devicename("CECExample");
        devicename.copy(cec_config.strDeviceName, std::min(devicename.size(),13u) );
        
        cec_config.clientVersion       = CEC::LIBCEC_VERSION_CURRENT;
        cec_config.bActivateSource     = 0;
        cec_config.callbacks           = &cec_callbacks;
        cec_config.deviceTypes.Add(CEC::CEC_DEVICE_TYPE_RECORDING_DEVICE);
    
        cec_callbacks.CBCecKeyPress    = &on_keypress;
    
        // Get a cec adapter by initialising the cec library
        CEC::ICECAdapter* cec_adapter = LibCecInitialise(&cec_config);
        if( !cec_adapter )
        { 
            std::cerr << "Failed loading libcec.so\n"; 
            return 1; 
        }
    
        // Try to automatically determine the CEC devices 
        CEC::cec_adapter devices[10];
        int8_t devices_found = cec_adapter->FindAdapters(devices, 10, NULL);
        if( devices_found <= 0)
        {
            std::cerr << "Could not automatically determine the cec adapter devices\n";
            UnloadLibCec(cec_adapter);
            return 1;
        }
    
        // Open a connection to the zeroth CEC device
        if( !cec_adapter->Open(devices[0].comm) )
        {        
            std::cerr << "Failed to open the CEC device on port " << devices[0].comm << std::endl;
            UnloadLibCec(cec_adapter);
            return 1;
        }
    
        // Loop until ctrl-C occurs
        while( !exit_now )
        {
            // nothing to do.  All happens in the CEC callback on another thread
            sleep(1);
        }
    
        // Close down and cleanup
        cec_adapter->Close();
        UnloadLibCec(cec_adapter);
    
        return 0;
    }

It does not compile using libcec4 and libcec4-dev and throws these errors ::

    cec-simplest.cpp: In function ‘int main(int, char**)’:
    cec-simplest.cpp:76:19: error: ‘CEC::ICECCallbacks {aka struct CEC::ICECCallbacks}’ has no member named ‘CBCecKeyPress’; did you mean ‘keyPress’?
         cec_callbacks.CBCecKeyPress    = &on_keypress;
                       ^~~~~~~~~~~~~
    cec-simplest.cpp:88:41: error: ‘class CEC::ICECAdapter’ has no member named ‘FindAdapters’; did you mean ‘PingAdapter’?
         int8_t devices_found = cec_adapter->FindAdapters(devices, 10, NULL);

When I renamed `CBCecKeyPress` to `keyPress` and `FindAdapters` to `PingAdapter` , I got these errors ::

    cec-simplest.cpp: In function ‘int main(int, char**)’:
    cec-simplest.cpp:76:33: error: invalid conversion from ‘int (*)(void*, CEC::cec_keypress)’ to ‘void (*)(void*, const cec_keypress*) {aka void (*)(void*, const CEC::cec_keypress*)}’ [-fpermissive]
         cec_callbacks.keyPress    = &on_keypress;
                                     ^~~~~~~~~~~~
    cec-simplest.cpp:88:70: error: no matching function for call to ‘CEC::ICECAdapter::PingAdapter(CEC::cec_adapter [10], int, NULL)’
         int8_t devices_found = cec_adapter->PingAdapter(devices, 10, NULL);
                                                                          ^
    In file included from cec-simplest.cpp:5:0:
    /usr/include/libcec/cec.h:77:18: note: candidate: virtual bool CEC::ICECAdapter::PingAdapter()
         virtual bool PingAdapter(void) = 0;
                      ^~~~~~~~~~~
    /usr/include/libcec/cec.h:77:18: note:   candidate expects 0 arguments, 3 provided

Some Info that I got about `keyPress` from `/usr/include/libcec/cectypes.h`  ::

    typedef struct cec_keypress
    {
      cec_user_control_code keycode;   /**< the keycode */
      unsigned int          duration;  /**< the duration of the keypress */
    } cec_keypress;
     
    
    typedef struct ICECCallbacks
    {
    void (CEC_CDECL* keyPress)(void* cbparam, const cec_keypress* key);
    
      /*!
       * @brief Transfer a CEC command from libCEC to the client.
       * @param cbparam             Callback parameter provided when the callbacks were set up
       * @param command             The command to transfer.
       */
     
      void Clear(void)
      {
        keyPress             = nullptr;

There is no documentation available for libcec.

What modifications do I need to do to make it work with libcec4 ?

opdenkamp, 2018-03-15:
Sorry, I can't help you with other people's code. If you need an example, see https://github.com/Pulse-Eight/libcec/blob/master/src/cec-client/cec-client.cpp

closed 2018-03-15 by opdenkamp

DrGeoff, 2018-10-07:
I'm the author of the code you are asking about. I know that a year has passed since you asked the question but I only stumbled onto this question because I was searching for the solution myself! LOL. Luckily I've cracked the problem by myself.

I've updated the github code to work https://github.com/DrGeoff/cec_simplest and I've written a blog post about the code https://drgeoffathome.wordpress.com/2018/10/07/a-simple-libcec4-example-for-the-raspberry-pi/

In short, these are the changes I had to make. First up, the _on_keypress_ function now is passed a pointer to a _cec_keypress_ message rather than a by-value copy of a message. The next change is that the CEC framework has changed the name of the callback function from _CBCecKeyPress_ to the simple _keyPress_. In a similar vein, the _FindAdapters_ function is now _DetectAdapters_ (not PingAdapters as you tried). And finally, the _DetectAdapters_ function fills in an array of _cec_adapter_descriptor_ rather than _cec_adapter_, which has the flow on effect to the _Open_ call taking a _strComName_ rather than simply _comm_.