Pulse-Eight/libcec #391: Submodules don't work well with release tar balls

MilhouseVH, 2017-11-21:
If you look at your release tarballs, you'll notice they're missing the submodules:
```
support @ 63ec40a
dotnet @ 180cbe2
platform @ 6535e48
```
Submodules don't work with release source tarballs, or snapshots.

Not sure if you can find another solution that doesn't use submodules, as they are the devils spawn. :)

opdenkamp, 2018-03-15:
Yeah I'm not a big fan of them either, but I don't have a better solution that doesn't involve copying or platform specific scripts, so they'll have to do. This is really a missing feature on Github imo.

opdenkamp, 2018-11-05:
Closing this one. If someone comes up with a better solution, then PRs are very welcome.

closed 2018-11-05 by opdenkamp

CvH, 2018-11-05:
@opdenkamp you can upload manual created releases to github

[image]

opdenkamp, 2018-11-05:
I'm aware of that, but there's no need to upload those tarballs to Github manually imo.

git-archive does not include submodules, so if you need an archive that does include submodules, then you can't use these Github's generated tarballs (which uses git-archive), but you need to create your own tarball which does include the things that you want included and upload it somewhere. For open/libreelec you can just use xbmc's mirrors for that, because I always upload them there after each release.