Pulse-Eight/libcec #393: Disable automatic hdmi switch to raspberry pi after lg tv power on

felipesalomao, 2017-12-29:
Hi, i am running raspbian, aways after i power lg tv, about 5 seconds later it switch to raspberry pi hdmi.. Usually tv start on sat hdmi imput.. I want to change that.. But i dont know how.. I already stoped Kodi to see if is some config is making that.. but even closed, it happens same.. But i never installed libcec on my raspberry, kodi could installed automatically, i dont know.. I even dont know where is libcec instalation path.. 
I found similar problem/solution [here](https://github.com/Pulse-Eight/libcec/issues/307#issuecomment-298149513)
But was running OSMC.. I run raspbian

What exactly i need to make/change using ssh / putty command line?

ValorNaram, 2018-01-29:
The Raspbian kernel makes that happen, not kodi

ValorNaram, 2018-01-29:
Try adding `hdmi_ignore_cec_init=1` to `/boot/config.txt`

opdenkamp, 2018-03-15:
This is not caused by libCEC if it happens on the Raspberry Pi without Kodi running. @ValorNaram's solution in https://github.com/Pulse-Eight/libcec/issues/393#issuecomment-361270289 should stop this indeed.

closed 2018-03-15 by opdenkamp

jammi, 2018-07-23:
This is caused by libCEC and the same as #307 which I'm experiencing as well after OpenElec upgraded the bundled libCEC version to 4.0.2.