Pulse-Eight/libcec #394: python binding does not work

ValorNaram, 2018-01-07:
Unfornately, the python binding for libcec does not work.

I do

    import cec
    cecconfig = cec.libcec_configuration()

but it says `AttributeError: 'module' object has no attribute 'libcec_configuration'`

JulianMined, 2018-01-12:
You are using the wrong library.
https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md

ValorNaram, 2018-01-12:
While compilation I got the following error:

    cmake -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..
    -- git found: libcec-4.0.2+30-8adc786
    CMake Error at src/libcec/cmake/CheckPlatformSupport.cmake:193 (if):
        if given arguments:

        "3" "GREATER" "2" "AND" "3" "GREATER_EQUAL" "7"

        Unknown arguments specified
    Call Stack (most recent call first):
        src/libcec/CMakeLists.txt:133 (include)

    -- Configuring incomplete, errors occurred!
    See also "/home/pi/libcec/build/CMakeFiles/CMakeOutput.log".
    See also "/home/pi/libcec/build/CMakeFiles/CMakeError.log".

I've done this previously and then I jumped to the official libcec package shipped with the debian repo. Now I am back compiling from source code and both methods does not work

michaelarnauts, 2018-02-02:
I have the same issue. A `4.0.2.1~raspbian_armhf` compiled from source works, but the one in the repositories is `4.0.2.2~jessie` and that doesn't work.

```
'module' object has no attribute 'libcec_configuration'
```

ValorNaram, 2018-02-02:
where did you get the source code for compilation

michaelarnauts, 2018-02-02:
You can find instructions by combining these two articles:
* https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md
* https://github.com/Pulse-Eight/libcec/blob/master/docs/README.debian.md

I had to make this change to libcec:
```diff
diff --git a/debian/rules b/debian/rules
index feb4782..7959436 100755
--- a/debian/rules
+++ b/debian/rules
@@ -8,4 +8,4 @@

 override_dh_auto_configure:
-       cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=1 -DCMAKE_INSTALL_PREFIX=/usr
+       cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=1 -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib -DCMAKE_INSTALL_PREFIX=/usr
```

opdenkamp, 2018-03-15:
The Debian packaging script is indeed meant for the PC and doesn't include the Raspberry Pi specific build options. I'll include the default options in the next version

closed 2018-03-15 by opdenkamp