Pulse-Eight/libcec #395: no sound after htpc (kodi 17) goes into suspend / no wake up when power on

blahmutz, 2018-01-07:
i have a marantz sr 7011 connected to my HTPC with a pulse eight CEC usb adapter. When i use the TV control and turn off the TV, the marantz turns off and the HTPC goes into suspend. When i power on the TV, the marantz wakes up but the HTPC not. This is working without the pulseeight cec adapter (turn on/off all devices and having sound with my tablet).

my TV is a sony bravia from 2016.

opdenkamp, 2018-03-15:
I've tried this with a 2015 Sony and I can't reproduce this issue.
The TV sends out a routing change right after it powers up on my Sony, before it polls the AVR and powers on the AVR. This instructs the PC to wake up from suspend.

Couple of questions:
- Did you wait at least 1 minute after the TV powered off before you tried to power it on? It could take up to a minute before it works.
- If you did and it doesn't work, can you wake up the PC by hitting a key on your (USB) keyboard or mouse? If not, then waking up the PC by USB is not enabled in your BIOS or operating system
- If you can wake up the PC with your mouse or keyboard, can you send me the output of cec-client in which you type the command `as`. Then power off the TV with the remote, wait ~30s, then power it on again, and send the full output so I can see what the TV is trying to do, which firmware the adapter is running, which OS, etc.

blahmutz, 2018-03-22:
1) yes
2) i can wake up by hitting the mouse, any key on the keyboard or by hitting the power button (atm i hit the power button as i dont want to have a keyboard on the htpc)
3) 
`root@htpc:/home/oli# cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.2
no serial port given. trying autodetect: 
 path:     /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3
 com port: /dev/ttyACM0

opening a connection to the CEC adapter...
DEBUG:   [              55]	Broadcast (F): osd name set to 'Broadcast'
ERROR:   [            3560]	error opening serial port '/dev/ttyACM0': Couldn't lock the serial port
ERROR:   [            3560]	could not open a connection (try 1)
ERROR:   [            6396]	error opening serial port '/dev/ttyACM0': Couldn't lock the serial port
ERROR:   [            6396]	could not open a connection (try 2)
ERROR:   [            8285]	error opening serial port '/dev/ttyACM0': Couldn't lock the serial port
ERROR:   [            8285]	could not open a connection (try 3)
ERROR:   [            9545]	error opening serial port '/dev/ttyACM0': Couldn't lock the serial port
ERROR:   [            9545]	could not open a connection (try 4)
unable to open the device on port /dev/ttyACM0
NOTICE:  [           10550]	connection opened
ERROR:   [           10550]	could not start CEC communications
root@htpc:/home/oli# su - kodi
kodi@htpc:~$ cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.2
no serial port given. trying autodetect: 
 path:     /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3
 com port: /dev/ttyACM0

opening a connection to the CEC adapter...
DEBUG:   [              56]	Broadcast (F): osd name set to 'Broadcast'
ERROR:   [            3558]	error opening serial port '/dev/ttyACM0': Device or resource busy
ERROR:   [            3558]	could not open a connection (try 1)
ERROR:   [            6394]	error opening serial port '/dev/ttyACM0': Device or resource busy
ERROR:   [            6394]	could not open a connection (try 2)
ERROR:   [            8284]	error opening serial port '/dev/ttyACM0': Device or resource busy
ERROR:   [            8284]	could not open a connection (try 3)
ERROR:   [            9544]	error opening serial port '/dev/ttyACM0': Device or resource busy
ERROR:   [            9544]	could not open a connection (try 4)
unable to open the device on port /dev/ttyACM0
NOTICE:  [           10549]	connection opened
ERROR:   [           10549]	could not start CEC communications
`

opdenkamp, 2018-11-02:
github user deleted, so no way to communicate about this. closing

closed 2018-11-02 by opdenkamp