Pulse-Eight/libcec #400: nothing but play/pause/fwd/back being passed through once soundbar added to newish panasonic tv

twelvedogs, 2018-01-23:
tv is a panasonic TH-55DX600U
sound bar is  panasonic SB-HWA488

not even sure if it's sanely fixable but the tv stops sending anything but fwd/back/play etc once it recognises the sound bar is there and inputs have been switched once, i'd provide logs but there's not much going on except the buttons that come through come through and those that don't don't

willing to provide any logs or i'll just remap the media buttons to something useful, all good

9000h, 2018-01-23:
you may have to read the readme https://github.com/Pulse-Eight/libcec#panasonic

twelvedogs, 2018-01-23:
sorry *only* the play pause etc work, the rest of the remote is not sent through

coder9, 2022-03-05:
Were you able to find a solution? I have the same problem, and I agree that it is the opposite of the issue explained in the readme, i.e. only the play/pause/stop/rew/ffwd buttons grouped at the bottom of the remote are passed through to Kodi, no other button is.

I power everything off, turn the TV on, followed by the soundbar, then the Raspberry Pi. All buttons work for a while, then the non-"cassette" buttons aren't passed through any more. Kodi debug logs show only the cassette buttons are detected and other buttons don't reach Kodi.

The TV is a Panasonic TX40-DX600E (2016) updated to the latest firmware (3280) - https://av.jpn.support.panasonic.com/support/global/cs/tv/download/fw/eu_2016/down_eu_uk_cis_02.html - and a Panasonic SC-HTB200 soundbar which doesn't appear to have any update available for it.

Hardware: Pi 4 4GB Model B Rev 1.2
OS: OSMC 2021-12
Kodi 19.3
Kernel: 5.10.78-2-osmc
cec-client: 6.0.2

twelvedogs, 2022-03-06:
as far as i can tell the TV no longer allows the buttons to be sent through to the pi once it realises the sound bar is plugged in, possibly because it thinks the bar is a player type device and might handle the remote buttons itself (just guessing).  i bought a flirc usb and use another remote

twelvedogs, 2022-03-06:
actually there is one other way around it, connect the sound bar via optical instead of ARC but then it loses CEC

coder9, 2022-03-08:
Thank you for replying. I'm using the first workaround you suggested, but working CEC would be much better. Could I ask you to re-open this issue so maybe the developers could take a look at it?

I've attached a Kodi debug log, with debug mode enabled and component-specific logging set to the CEC library and the place where it lost button input commented.
 
[kodi.log.txt](https://github.com/Pulse-Eight/libcec/files/8210332/kodi.log.txt)

twelvedogs, 2022-03-10:
good luck

closed 2025-02-27 by github-actions[bot]