Pulse-Eight/libcec #402: Help understanding libcec client addressing (why am I always getting f.f.f.f address?)

trinitronx, 2018-01-26:
Originally posted as comment to #319 (sorry for the re-post... I just wanted to get a little assistance understanding what is going on here)

I am seeing a very weird issue with my client code always getting the `f.f.f.f` address, using Pulse8 CEC USB adapter.  I **can reproduce** using a custom built `golang` CEC client.  I also can test with standard `cec-client`, but I get a duplicate `1.0.0.0` address.  So there are some weird addressing issues with either client?

The testing host running this code is a plain Linux box with Pulse8 CEC USB adapter attached.  Also attached to the same TV is a Raspberry Pi running Kodi, and Sony Bluray player.

I have some very "rough around the edges" [golang code][1] which can reproduce this issue in my environment.  It uses [a CGO binding][2] into official `libcec` from OS package (Ubuntu Artful `17.10`: `libcec4 4.0.1+dfsg1-1`, Linked against: `libcec-dev 4.0.1+dfsg1-1`).

Seems that after running the Golang code binary `test-cec` to transmit some CEC commands, it results in adding itself as a device with physical address `f.f.f.f`.  (Does this address even make sense??  Is this like broadcast address in TCP/IP?  Maybe this is a problem with the way that the Golang code does [`Open()`][3] [`libcec_initialise()`][4], [`getAdapter()` or `openAdapter()`][5]

I can then run `echo 'scan' | cec-client  -s`  to show that this fake device from Golang code as client is replaced by official `cec-client` (`osd string:    CECTester`).  `cec-client` seems to fix itself onto **`1.0.0.0`** address (which is already (**in use** by my Raspberry Pi).  **However**, the weird part is that the device running Golang code and subsequent `cec-client` is actually on **HDMI-4**, which should be **`4.0.0.0`** *correct*?

While playing around with this Golang code & sending the "power on" and "set active source" commands using various source & dest logical address, I somehow got things in a state where the CEC commands sent from TV remote would not be received anymore by Kodi on Raspberry Pi, and MythTV on same box as HDMI-4.  This could be some problem with the remote itself, as I usually can unplug batteries and re-plug them in and it works again.  This time it didn't.     I could then unplug TV & plug it back in... then any `f.f.f.f` device and `1.0.0.0` goes away and it is set correctly back to `4.0.0.0`.  Devices again responded.   Not sure what was going on there or how to debug it.

Either way, there's some weirdness with addressing going on... not sure how the CEC standard says this is supposed to be derived if I'm seeing `f.f.f.f` and duplicate `1.0.0.0` for the same device using my client & official `cec-client`.

@opdenkamp :  Is this something that I could ask you or another `libcec` developer to take a quick look at just to sanity check my example to see if everything looks right?  Maybe it's something simple or stupid I'm missing.  I'm a newbie to this CEC protocol & golang in general :-)

[-->Logs for both commands are here<--][6]

[1]: https://github.com/trinitronx/test-cec
[2]: https://github.com/chbmuc/cec/blob/master/libcec.go
[3]: https://github.com/chbmuc/cec/blob/master/cec.go#L58-L82
[4]: https://github.com/chbmuc/cec/blob/master/libcec.go#L53-L69
[5]: https://github.com/chbmuc/cec/blob/master/libcec.go#L71-L88
[6]: https://gist.github.com/trinitronx/cb96187c00b5077e44ea165405d078d8

ndarilek, 2018-08-17:
Did you ever figure this out? I'm in the same situation.

trinitronx, 2018-08-29:
@ndarilek : No, I ended up getting [one of these remotes](https://www.mythtv.org/wiki/Air_mouse_rf_remote) instead of relying on CEC control

opdenkamp, 2020-07-10:
very late response, but the address is now detected correctly from EDID in the recent versions, but this will only work while X is running. when running without X or if cec-client doesn't have access, you have to specify the correct HDMI input with the ` -p` command line switch.

F.F.F.F means invalid/not detected

closed 2020-07-10 by opdenkamp

trinitronx, 2020-07-20:
@opdenkamp Thanks for the helpful explanation!  I have recently done a fresh OS install, so I'll put this on my list of things to test out again 😄