Pulse-Eight/libcec #403: AVR keeps turning on after turning off TV

zedtools, 2018-02-03:
I have an issue when I connect an Apple TV 4K to my setup (RPi 3 running Kodi, Sony TV, Onkyo TX-NR414 AVR). Whenever Kodi is the active source, and I turn off my TV, the AVR turns off as expected, but a few seconds later the AVR turns back on.

I managed to track it down to the following behaviour:

1. When I turn off the TV, the Apple TK 4K sends a "requests active source" command
2. Kodi response with "broadcast (F): active source (4300)"
3. I am fairly certain this is what is causing the AVR to turn back on

Here is a cec-client log capturing this behaviour: [TV-off-AVR-on.log](https://github.com/Pulse-Eight/libcec/files/1691383/TV-off-AVR-on.log)

If Kodi is not the active source, and I turn off the TV, the AVR stays off.

I managed to compile libCEC on my RPi 3 with the following patch, and this appears to fix the issue:

```diff
--- a/src/libcec/implementations/CECCommandHandler.cpp
+++ b/src/libcec/implementations/CECCommandHandler.cpp
@@ -37,6 +37,7 @@
 #include "devices/CECBusDevice.h"
 #include "devices/CECAudioSystem.h"
 #include "devices/CECPlaybackDevice.h"
+#include "devices/CECTV.h"
 #include "CECClient.h"
 #include "CECProcessor.h"
 #include "LibCEC.h"
@@ -548,9 +549,17 @@ int CCECCommandHandler::HandleRequestActiveSource(const cec_command &command)
     LIB_CEC->AddLog(CEC_LOG_DEBUG, ">> %i requests active source", (uint8_t) command.initiator);
     m_processor->GetDevice(command.initiator)->SetPowerStatus(CEC_POWER_STATUS_ON);
 
-    std::vector<CCECBusDevice *> devices;
-    for (size_t iDevicePtr = 0; iDevicePtr < GetMyDevices(devices); iDevicePtr++)
-      devices[iDevicePtr]->TransmitActiveSource(true);
+    // Do not transmit active source if TV is off, to prevent AVR from turning back on
+    // after TV is powered off
+    cec_power_status tv_state = m_processor->GetTV()->GetCurrentPowerStatus();
+    LIB_CEC->AddLog(CEC_LOG_DEBUG, "Got TV power status: %i", (uint8_t) tv_state);
+
+    if (tv_state == CEC_POWER_STATUS_ON)
+    {
+      std::vector<CCECBusDevice *> devices;
+      for (size_t iDevicePtr = 0; iDevicePtr < GetMyDevices(devices); iDevicePtr++)
+        devices[iDevicePtr]->TransmitActiveSource(true);
+    }
   }
 
   return COMMAND_HANDLED;
```

There are more details on this forum post:

https://forum.kodi.tv/showthread.php?tid=327971

I don't know whether the fault lies with the Apple TV 4K, my AVR, or libCEC. Is this something that can be fixed in libCEC?

jammi, 2018-07-23:
I have this same exact issue. Since I don't have a build setup on my raspi, could you supply the compiled files in question to test them out?

zedtools, 2018-07-23:
@jammi File attached here.

[libcec.so.4.0.2.gz](https://github.com/Pulse-Eight/libcec/files/2221460/libcec.so.4.0.2.gz)

If you are using OSMC, just copy the file to /usr/osmc/lib/libcec.so.4.0.2 and restart OSMC. I have not tested this file with other distributions of Kodi,

Portisch, 2020-05-02:
@opdenkamp we got a new report because of this issue. It looks like this fix from @zedtools is needed. Can you please have look? Thanks!
https://discourse.coreelec.org/t/avr-keeps-turning-on-after-turning-off-tv-when-apple-tv-4k-is-connected/10962

closed 2025-02-27 by github-actions[bot]