Pulse-Eight/libcec #405: CEC connection lost when TV goes into standby or has been switched off

radiowest, 2018-02-21:
## Hardware

- LG Television
- Raspberry Pi 3

## Software
- `libcec4/stable,now 4.0.2.3~stretch armhf [installed,automatic]`
- `kodi/stable,now 2:17.6-2~stretch all [installed]`

## Bug Description

This occurs with a Raspberry Pi 3 connected to a LG television. The television also has another set-top box connected which provides digital television.

If the television is switched ON before being connected CEC works as intended. We are able to switch inputs between our raspberry pi and the set-top box, and back again, without losing CEC.

When the TV goes into standby mode or is switched off, and then switched on again, CEC connection is lost between the raspberry pi and the television.

## Workaround

CEC connection can be restored through the following sequence

1. Switch **OFF** the set-top box
2. Select Raspberry Pi HDMI output on the television
3. Open LG **SIMPLINK** menu and reselect *HDD RECORDER* option which is the HDMI cable connected to the Raspberry Pi
4. Repeat step 3 two times

## Notes

This behaviour **also occurs** with the set-top box physically disconnected from the TV

radiowest, 2018-02-24:
## Update

CEC connection is lost **but can be restored** if the TV is switched ON with the Raspberry Pi HDMI input selected. If the TV was switched OFF while on the set-top box HDMI input it is not possible to restore CEC even after switching to the Raspberry Pi input.

## Conclusion

In all cases CEC is lost when the TV has cycle power or standby mode. But it can be restored by switched the TV ON with the Raspberry Pi HDMI input selected.

johnmanko, 2021-03-10:
I'm running Kodi on linux, and when I turn off/on the television (LG), Kodi no longer receives CEC input.  Either the Pulse-Eight hardware is defective (doubt it), Kodi is defective (maybe), or the libcec is failing.  It could be the tv, but restarting Kodi resolves the issue, which tells me its a software, not hardware, issue.

If you instruct me how to obtain debugging logs for libcec I will provide them.

closed 2025-02-27 by github-actions[bot]