Pulse-Eight/libcec #413: Adapter identification

kdschlosser, 2018-04-15:
I am looking for a way to identify an adapter via the library. something like the serial number. It has to be a unique identifier. I am working on the EventGhost plugin and one thing i noticed can and i am sure will be an issue is there is no way to identify a specific adapter other then by it's com port. Now this is not a good way because a com port can change. simply by plugging the adapter into a different USB port. If i want to retain the settings inside of EventGhost for a specific adapter you can see how this would cause an issue. 

Since I am writing this plugin from scratch again if there is a way to go about this and someone could shed some light on it that would be a huge help.

Thanks again.

opdenkamp, 2018-11-02:
I've looked into adding something to do this earlier today, but we have no ROM space available to add anything like this in the current version of the CEC adapter. You can only identify it by it's location (USB port/hub)

closed 2018-11-02 by opdenkamp

kdschlosser, 2018-11-02:
OOOoO man.... that hurts.. There is no ESN on the thing that can be pulled?? I find that rather strange with current technology. Just about anything that is electronic has an ESN in it that can be pulled from it. 

be it as it may. kinda depressing. Windows is a bugger with the serial ports. and installing a new driver if you simply move the device from one port to another. which then creates a new GUID. so it cannot be tracked that way. I could have sworn that in order to register a driver in windows for the device it requires some form of a serial number. I can't remember. and I do not know if this holds true with comports.

opdenkamp, 2018-11-05:
The adapter does not provide any unique ID, and it's not possible to add one in this model unfortunately.

kdschlosser, 2018-11-06:
That's cool. some food for thought for the next model.