Pulse-Eight/libcec #423: Raspberry Pi 3B+ Hassbian Segfault

jacobschaer, 2018-06-19:
I built libcec and platform from source following [README.raspberrypi.md](https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md) (except I added `-DCMAKE_BUILD_TYPE=Debug` where appropriate). I have my Raspberry Pi connected to either my Nintendo Switch or Sony AV Receiver (both should have CEC enabled) using an HDMI cable that supports CEC. I also enabled `hdmi_force_hotplug=1` to `/boot/config.txt` following some earlier unsuccessful attempts. Now when I try to discover devices on the network it runs for a while and eventually 

stdout/stderr (with full diagnostics on)
[log.txt](https://github.com/Pulse-Eight/libcec/files/2114147/log.txt)

Backtrace from coredump:
```
Core was generated by `cec-client -s -d 1'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x76b84f0c in CEC::CRPiCECAdapterMessageQueue::MessageReceived (
    this=0x1c71428, opcode=CEC::CEC_OPCODE_REQUEST_ACTIVE_SOURCE,
    initiator=CEC::CECDEVICE_RECORDINGDEVICE1,
    destination=CEC::CECDEVICE_UNREGISTERED, response=0)
    at /home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterMessageQueue.cpp:124
124         bHandled = it->second->MessageReceived(opcode, initiator, destination, response);
---Type <return> to continue, or q <return> to quit---
[Current thread is 1 (Thread 0x746ff450 (LWP 22537))]
(gdb) bt full
#0  0x76b84f0c in CEC::CRPiCECAdapterMessageQueue::MessageReceived (
    this=0x1c71428, opcode=CEC::CEC_OPCODE_REQUEST_ACTIVE_SOURCE,
    initiator=CEC::CECDEVICE_RECORDINGDEVICE1,
    destination=CEC::CECDEVICE_UNREGISTERED, response=0)
    at /home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterMessageQueue.cpp:124
        it = <error reading variable: Cannot access memory at address 0x61636967>
        bHandled = false
        lock = {<P8PLATFORM::PreventCopy> = {<No data fields>},
          m_mutex = @0x1c71430, m_bClearOnExit = false}
#1  0x76b832ac in CEC::CRPiCECAdapterCommunication::OnDataReceived (
    this=0x1c71388, header=131073, p0=34079, p1=0, p2=0, p3=0)
    at /home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.cpp:184
        reason = VC_CEC_TX
#2  0x76b82c60 in rpi_cec_callback (callback_data=0x1c71388, p0=131073,
    p1=34079, p2=0, p3=0, p4=0)
    at /home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.cpp:58
No locals.
#3  0x7695ccf4 in cecservice_notify_func () from /opt/vc/lib/libbcm_host.so
No symbol table info available.
#4  0x76990cc4 in vcos_thread_entry (arg=0x76976600 <cecservice_notify_task>)
    at /home/dc4/projects/staging/userland/interface/vcos/pthreads/vcos_pthreads---Type <return> to continue, or q <return> to quit---
.c:144
        ret = <optimized out>
        thread = 0x76976600 <cecservice_notify_task>
#5  0x76f48fc4 in start_thread (arg=0x746ff450) at pthread_create.c:335
        pd = 0x746ff450
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {-1764532302, -1807204738,
                1953494096, 2126263068, 0, 338, 2126263068, 4096, 1,
                1953492876, 0 <repeats 54 times>}, mask_was_saved = 0}},
          priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0,
              cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
Backtrace stopped: Cannot access memory at address 0x66

```

Does anything jump out from this? I can provide more details from the core file as needed.

jacobschaer, 2018-06-20:
I installed the latest full Raspbian and it works fine... so probably some driver issue in Hassbian or a different library version. I'll try to compare when I get some more time.

opdenkamp, 2018-11-02:
the queue was accessed without a lock, which caused this crash: one thread removing items while the other was iterating over it

closed 2018-11-02 by opdenkamp