Pulse-Eight/libcec #424: cec-client on Windows 10 --> TV turns on automatically after some minutes

TuxFloHB, 2018-06-21:
Hi, 

we are using the USB-CEC Adapter with Windows 10 and a Samsung MU6179 TV. 
Powering on and off the TV is working fine. 
But when the TV is powered off, after 10 - 120 minutes it always turns on again automatically!

The TV has already been replaced, because we thought that this was a hardware issue. But nothing changed. When I remove the cec adapter, the TV stays off.

Do you have any idea for these behaviour? 

regards
Florian

PeterW123, 2018-06-25:
We have exactly same issue with Windows 10 and Samsung MU6300 TV, the TV always turns on automatically by USB-CEC Adapter.  Just waiting for solution.

aanderse, 2018-09-18:
I have a Samsung UN50MU6070F which turns on randomly 2-5 times a day if my pulse 8 cec adapter is plugged in. This happens regardless of operating system (tested on Windows 10 and several Linux variants).

If there is any actions I could take, any logs I could produce to help resolve this issue please let me know.

closed 2018-11-02 by opdenkamp

opdenkamp, 2018-11-06:
it needs a new firmware which has been tested today and will be available later this week

opdenkamp, 2018-11-07:
you can download the firmware upgrade tool here: [cec-firmware-v9.zip](https://github.com/Pulse-Eight/libcec/files/2557238/cec-firmware-v9.zip)
it will be uploaded to our website

aanderse, 2018-11-07:
Great thanks! Will try out tonight.

TuxFloHB, 2018-11-11:
When will the new driver for Windows be available?

aanderse, 2018-11-12:
@opdenkamp recompiled libcec against d57d3ac91a2e6c7f224082fc749331210718a08b and then built kodi this morning... and unfortunately tv just turned on 20 minutes ago. Note that I have applied the firmware update you provided above.

After updating libcec and kodi I rebooted the pc, put the computer to sleep, and then the tv was off. Can you give me the steps you took to reproduce successfully keeping the tv off? Is there any logging information I can produce that might give us a clue?

aanderse, 2018-11-14:
@opdenkamp Whenever I enable cec logging in component debugging the setting will not persist across reboots or even restarting Kodi.

Attached is my kodi.log file.
[kodi.log](https://github.com/Pulse-Eight/libcec/files/2579326/kodi.log)

TuxFloHB, 2018-11-19:
@opdenkamp Where can I download the new driver for Windows to fix the issue with our Samsung TV?

aanderse, 2018-11-19:
@opdenkamp I've left my computer on for the last ~24 hours with the TV turned off and the TV has not turned on. So it appears that as long as the computer is turned on your fixes are working as expected, though if the computer is turned off or in sleep mode this doesn't work.

What is the expectation?

commit 2c41938e (2018-11-02): changed: disable autonomous mode when a Samsung TV is connected

Samsung 2017+ TVs will power on randomly (Samsung bug)

closes #424