Pulse-Eight/libcec #426: Problem with libcec on tv Toshiba

mparadiso78, 2018-07-09:
I have the Pulse Eight device running on Samsung Smart TV perfectly.
A few months ago, I bought a Toshiba Smart TV that uses the T-Link system.
Unfortunately the device does not work on this TV.
I have tried several solutions but so far I do not know how I can solve this problem. I got in touch with Toshiba and they say they do not know how to help me with this.
Is there anything to do?

malard, 2018-07-09:
can you capture logs from our adapter that shows traffic between our device and the TV? if so, email them or paste them here and we can look into it

mparadiso78, 2018-07-17:
HI,

Thank you for your help.
Below is the log with the errors that occurred.

DEBUG:                1738 connection opened, clearing any previous input and waiting for active transmissions to end before starting
DEBUG:                1739 communication thread started
DEBUG:                1750 turning controlled mode on
NOTICE:               1773 connection opened
DEBUG:                1774 processor thread started
DEBUG:                1774 << Broadcast (F) -> TV (0): POLL
TRAFFIC:              1774 << f0
DEBUG:                1774 setting the line timeout to 3
DEBUG:                1814 >> POLL sent
DEBUG:                1814 TV (0): device status changed into 'present'
DEBUG:                1814 << requesting vendor ID of 'TV' (0)
TRAFFIC:              1814 << f0:8c
TRAFFIC:              2041 >> 0f:87:f0:f0:f0:00
DEBUG:                2041 TV (0): vendor = Unknown (f0f0f0)
DEBUG:                2041 >> TV (0) -> Broadcast (F): device vendor id (87)
DEBUG:                2041 expected response received (87: device vendor id)
NOTICE:               2041 registering new CEC client - v4.0.2
DEBUG:                2041 detecting logical address for type 'recording device'
DEBUG:                2041 trying logical address 'Recorder 1'
DEBUG:                2041 << Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC:              2041 << 11
DEBUG:                2076 CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC:              2076 << 11
DEBUG:                2111 CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG:                2111 >> POLL not sent
DEBUG:                2111 using logical address 'Recorder 1'
DEBUG:                2111 Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:                2111 Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:                2111 Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:                2111 Recorder 1 (1): CEC version 1.4
DEBUG:                2111 CEC::CCECClient::AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:                2111 setting ackmask to  2
DEBUG:                2116 Recorder 1 (1): osd name set to 'CEC Tray'
DEBUG:                2116 Recorder 1 (1): menu language set to ''
DEBUG:                2116 CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - trying to get the physical address via ADL
DEBUG:                2118 CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - ADL returned physical address 0000
DEBUG:                2118 CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - trying to get the physical address via nvidia driver
DEBUG:                2118 CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - nvidia driver returned physical address 0000
DEBUG:                2118 CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - trying to get the physical address from the OS
DEBUG:                2118 CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - OS returned physical address 0000
NOTICE:               2118 setting HDMI port to 1 on device TV (0)
DEBUG:                2119 Recorder 1 (1): physical address changed from ffff to 1000
DEBUG:                2119 << Recorder 1 (1) -> broadcast (F): physical address 1000
TRAFFIC:              2119 << 1f:84:10:00:01
DEBUG:                2255 using persisted autonomous mode setting: 'enabled'
DEBUG:                2260 using persisted CEC version setting: '1.4'
DEBUG:                2266 using persisted logical address setting: 'Recorder 1'
DEBUG:                2272 using persisted device type setting: 'recording device'
DEBUG:                2278 using persisted logical address mask setting: '206'
DEBUG:                2294 using persisted device name setting: 'CEC Tray'
DEBUG:                2300 using persisted physical address setting: '1000'
DEBUG:                2300 making Recorder 1 (1) the active source
DEBUG:                2300 TV (0): power status changed from 'unknown' to 'in transition from standby to on'
DEBUG:                2300 marking the adapter as active source
NOTICE:               2305 >> source activated: Recorder 1 (1)
DEBUG:                2305 scheduling active source message for 'Recorder 1'
NOTICE:               2305 CEC client registered: libCEC version = 4.0.2, client version = 4.0.2, firmware version = 7, firmware build date: Wed Apr 13 10:36:11 2016 +0000, logical address(es) = Recorder 1 (1) , base device: TV (0), HDMI port number: 1, physical address: 1.0.0.0, compiled on Windows-6.3.9600 (x64), features: P8_USB, P8_detect
DEBUG:                2305 << Recorder 1 (1) -> TV (0): OSD name 'CEC Tray'
TRAFFIC:              2305 << 10:47:43:45:43:20:54:72:61:79
TRAFFIC:              2359 >> 01:46
DEBUG:                2359 >> TV (0) -> Recorder 1 (1): give osd name (46)
DEBUG:                2608 << Recorder 1 (1) -> TV (0): OSD name 'CEC Tray'
DEBUG:                2608 << requesting power status of 'TV' (0)
TRAFFIC:              2608 << 10:47:43:45:43:20:54:72:61:79
TRAFFIC:              2608 << 10:8f
[... 78 lines cut ...]
DEBUG:                3881 CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC:              3881 << 1c
DEBUG:                3914 CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG:                3914 >> POLL not sent
DEBUG:                3914 Reserved 1 (C): device status changed into 'not present'
DEBUG:                3914 << Recorder 1 (1) -> Reserved 2 (D): POLL
[... cut ...]
DEBUG:                4049 CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG:                4049 >> POLL not sent
DEBUG:                4049 Free use (E): device status changed into 'not present'
NOTICE:               4088 << powering on 'TV' (0)
TRAFFIC:              4088 << 10:04
NOTICE:               4143 << Recorder 1 (1) -> broadcast (F): active source (1000)
TRAFFIC:              4143 << 1f:82:10:00
DEBUG:                4251 << Recorder 1 (1) -> TV (0): menu state 'activated'
TRAFFIC:              4251 << 10:8e:00

mparadiso78, 2018-08-08:
Could someone help me with this problems?
Unfortunately I can no longer use the remote control with the pulse-eight because of this.

Thanks

opdenkamp, 2018-11-05:
Just looking into this issue. The problem is caused by an invalid vendor ID that's sent out by the TV (0xf0f0f0 instead of 0x008045). This ID is invalid, so the TV isn't recognised as Panasonic.

Check whether the firmware of the TV is up to date.

If there are no updates, then you can probably work around it by starting cec-client like this:
`cec-client -t p`

opdenkamp, 2018-11-06:
Tested master on a Toshiba now, and I'm not seeing any issues. Also tried with the work around that I mentioned, and that works too.

Because this is caused by a bug in your TV (no other reports from Panasonic owners about this), I can't fix this in code.

Passing `-p` as parameter to cec-client will work around it. If you're using Kodi, then you'll have to manually edit a configuration file to work around it:
* exit kodi
* open ~/.kodi/userdata/peripheral_data/usb_2548_1002_CEC_Adapter.xml
* change the line `<setting id="tv_vendor" value="0" />` into `<setting id="tv_vendor" value="32837" />`
* start kodi

closed 2018-11-06 by opdenkamp