Pulse-Eight/libcec #430: Using LibCecSharp.dll in C# Code

Danthekilla, 2018-08-09:
I'm trying to use the LibCecSharp.dll in one of my own C# applications.

Therefore I added the DLL into the references and I created some code based on the CecSharpTester which is located in the src folder. (I have also directly tried the CecSharpTester sample)

But all time I'm trying to create my project I get an error message:

`An unhandled exception of type 'System.IO.FileNotFoundException' occurred in Unknown Module.
Could not load file or assembly 'LibCecSharp.dll' or one of its dependencies. The specified module could not be found.`

I put the libcec.dll in the same directory as LibCecSharp.dll and my exe and the error still occurs.

alandurkin, 2018-10-01:
Hi @CrimsonCastel, did you find a resolution for this? I'm having the same problem.

opdenkamp, 2018-11-02:
you get this error when cec.dll can't be found in the path indeed. see https://github.com/Pulse-Eight/libcec/blob/master/docs/README.developers.md

closed 2018-11-02 by opdenkamp

fthmhmtozclk, 2019-10-11:
Try changing architecture to x86