Pulse-Eight/libcec #431: Problems with the pulse-eight on tv Toshiba TCL

mparadiso78, 2018-08-23:
I have the pulse-eight device and started using it on my new Toshiba TCL.
Unfortunately no command works on it and I have tried to solve this in several ways because in the old Samsung TV everything worked perfectly.
I will put the error log below but I am very sad because I am totally unable to use the remote control and I only use Kodi by android app.

Thank you

DEBUG: 1738 connection opened, clearing any previous input and waiting for active transmissions to end before starting
DEBUG: 1739 communication thread started
DEBUG: 1750 turning controlled mode on
NOTICE: 1773 connection opened
DEBUG: 1774 processor thread started
DEBUG: 1774 << Broadcast (F) -> TV (0): POLL
TRAFFIC: 1774 << f0
DEBUG: 1774 setting the line timeout to 3
DEBUG: 1814 >> POLL sent
DEBUG: 1814 TV (0): device status changed into 'present'
DEBUG: 1814 << requesting vendor ID of 'TV' (0)
TRAFFIC: 1814 << f0:8c
TRAFFIC: 2041 >> 0f:87:f0:f0:f0:00
DEBUG: 2041 TV (0): vendor = Unknown (f0f0f0)
DEBUG: 2041 >> TV (0) -> Broadcast (F): device vendor id (87)
DEBUG: 2041 expected response received (87: device vendor id)
NOTICE: 2041 registering new CEC client - v4.0.2
DEBUG: 2041 detecting logical address for type 'recording device'
DEBUG: 2041 trying logical address 'Recorder 1'
DEBUG: 2041 << Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: 2041 << 11
DEBUG: 2076 CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: 2076 << 11
DEBUG: 2111 CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG: 2111 >> POLL not sent
DEBUG: 2111 using logical address 'Recorder 1'
DEBUG: 2111 Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG: 2111 Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG: 2111 Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG: 2111 Recorder 1 (1): CEC version 1.4
DEBUG: 2111 CEC::CCECClient::AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG: 2111 setting ackmask to 2
DEBUG: 2116 Recorder 1 (1): osd name set to 'CEC Tray'
DEBUG: 2116 Recorder 1 (1): menu language set to ''
DEBUG: 2116 CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - trying to get the physical address via ADL
DEBUG: 2118 CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - ADL returned physical address 0000
DEBUG: 2118 CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - trying to get the physical address via nvidia driver
DEBUG: 2118 CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - nvidia driver returned physical address 0000
DEBUG: 2118 CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - trying to get the physical address from the OS
DEBUG: 2118 CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - OS returned physical address 0000
NOTICE: 2118 setting HDMI port to 1 on device TV (0)
DEBUG: 2119 Recorder 1 (1): physical address changed from ffff to 1000
DEBUG: 2119 << Recorder 1 (1) -> broadcast (F): physical address 1000
TRAFFIC: 2119 << 1f:84:10:00:01
DEBUG: 2255 using persisted autonomous mode setting: 'enabled'
DEBUG: 2260 using persisted CEC version setting: '1.4'
DEBUG: 2266 using persisted logical address setting: 'Recorder 1'
DEBUG: 2272 using persisted device type setting: 'recording device'
DEBUG: 2278 using persisted logical address mask setting: '206'
DEBUG: 2294 using persisted device name setting: 'CEC Tray'
DEBUG: 2300 using persisted physical address setting: '1000'
DEBUG: 2300 making Recorder 1 (1) the active source
DEBUG: 2300 TV (0): power status changed from 'unknown' to 'in transition from standby to on'
DEBUG: 2300 marking the adapter as active source
NOTICE: 2305 >> source activated: Recorder 1 (1)
DEBUG: 2305 scheduling active source message for 'Recorder 1'
NOTICE: 2305 CEC client registered: libCEC version = 4.0.2, client version = 4.0.2, firmware version = 7, firmware build date: Wed Apr 13 10:36:11 2016 +0000, logical address(es) = Recorder 1 (1) , base device: TV (0), HDMI port number: 1, physical address: 1.0.0.0, compiled on Windows-6.3.9600 (x64), features: P8_USB, P8_detect
DEBUG: 2305 << Recorder 1 (1) -> TV (0): OSD name 'CEC Tray'
TRAFFIC: 2305 << 10:47:43:45:43:20:54:72:61:79
TRAFFIC: 2359 >> 01:46
DEBUG: 2359 >> TV (0) -> Recorder 1 (1): give osd name (46)
DEBUG: 2608 << Recorder 1 (1) -> TV (0): OSD name 'CEC Tray'
DEBUG: 2608 << requesting power status of 'TV' (0)
TRAFFIC: 2608 << 10:47:43:45:43:20:54:72:61:79
[... 79 lines cut ...]
DEBUG: 3881 CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: 3881 << 1c
DEBUG: 3914 CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG: 3914 >> POLL not sent
DEBUG: 3914 Reserved 1 (C): device status changed into 'not present'
DEBUG: 3914 << Recorder 1 (1) -> Reserved 2 (D): POLL
TRAFFIC: 3914 << 1d
DEBUG: 3948 CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: 3948 << 1d
DEBUG: 3981 CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG: 3981 >> POLL not sent
DEBUG: 3981 Reserved 2 (D): device status changed into 'not present'
DEBUG: 3981 << Recorder 1 (1) -> Free use (E): POLL
TRAFFIC: 3981 << 1e
DEBUG: 4015 CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: 4015 << 1e
DEBUG: 4049 CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG: 4049 >> POLL not sent
DEBUG: 4049 Free use (E): device status changed into 'not present'
NOTICE: 4088 << powering on 'TV' (0)
TRAFFIC: 4088 << 10:04
NOTICE: 4143 << Recorder 1 (1) -> broadcast (F): active source (1000)
TRAFFIC: 4143 << 1f:82:10:00
DEBUG: 4251 << Recorder 1 (1) -> TV (0): menu state 'activated'
TRAFFIC: 4251 << 10:8e:00

konistehrad, 2018-09-13:
Seconding this with a Panasonic via a Monoprice CEC switch. I see almost the same log.

I'm on Windows 10 (latest), with an Nvidia 1070. Any other info I can provide please let me know.

opdenkamp, 2018-11-06:
dupe of #426 

re panasonic: please open a new ticket if it still happens on master, and attach a debug log. thanks

closed 2018-11-06 by opdenkamp