Pulse-Eight/libcec #441: currently active source: unknown (-1) when RPI is active source

potatowave, 2018-11-16:
On a raspberry pi 3 running Raspbian with libCEC 4.02 plugged into a 2018 Samsung Smart TV.

`echo "scan" | cec-client -d 1 -s RPI` returns

```
log level set to 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Samsung
osd string:    TV
CEC version:   1.4
power status:  on
language:      eng
 
 
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng
 
 
currently active source: unknown (-1)
```
If I change input to anything else, output is the same except for:

`currently active source: TV(0)`

Questions:

1.  Is there a way to get 'currently active source' reliably or is this a hardware/tv quirk?
2.  What I'm actually interested in is knowing which source/input is currently active on the screen.  Can I get this information?  (E.g current source is HDMI1)

andyjjones28, 2019-08-04:
+1 Suffering the exact same issue.

WalidAyyoub, 2019-11-10:
Were you able to solve this? I am facing the same issue, I am able to get the active source reported for TV, or for my PS4, but I get the unknown code when my device HDMI channel is active, by looking at the debug, it seems like the TV never respond back to the active source code only when the device running LibCEC source is active, which is strange.

cakechris, 2019-11-14:
+1 I'm seeing the same problem.

Chief90, 2021-08-08:
I have the same issue..

Jibun-no-Kage, 2025-02-26:
Cannot test with CEC 7.0... not available...
```
Get:1 http://deb.debian.org/debian bookworm/main arm64 libp8-platform2 arm64 2.1.0.1+dfsg1-4 [20.9 kB]
Get:2 http://archive.raspberrypi.com/debian bookworm/main arm64 libcec6 arm64 6.0.2-5+rpt2 [188 kB]
Get:3 http://archive.raspberrypi.com/debian bookworm/main arm64 cec-utils arm64 6.0.2-5+rpt2 [30.0 kB]
```

Jibun-no-Kage, 2025-02-26:
Seems that original scan error fixed overall with Debian 12 based PiOS release?
```
# echo 'scan' | cec-client -d 1
log level set to 1
No device type given. Using 'recording device'
CEC Parser created - libCEC version 6.0.2
no serial port given. trying autodetect:
 path:     /sys/devices/platform/soc/3f902000.hdmi/cec0
 com port: /dev/cec0

opening a connection to the CEC adapter...
waiting for input
requesting CEC bus information ...
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
```
But the active source issue remains.

opdenkamp, 2025-02-27:
active source works by sending a CEC query and if a device responds with "I am the active source", then that one is active.
if no response comes in after a timeout, then there is no (CEC capable) active source on the bus. this could be a non-CEC device, or a device that's not responding to the query correctly.

this is not a bug in libCEC, but this is how the CEC spec works.

closed 2025-02-27 by opdenkamp