Pulse-Eight/libcec #442: Inquiring about plans to merge 4.0.3 to release branch

mseeley, 2018-11-21:
👋 Hi @opdenkamp thanks a lot for your continued effort.

I wanted to ask about your plans to merge 4.0.3 to the [`release`](https://github.com/Pulse-Eight/libcec/tree/release) branch. Before asking I wasn't able to find any context about possibly deferring the merge as part of a staggered release. I hope you don't mind my asking. 👼 

Many thanks!

mseeley, 2018-11-26:
Hi @opdenkamp just one more kind bump to ask about `release` branch merge plans. We can work around on our own if you're otherwise unavailable.

Best regards, @mseeley. :)

opdenkamp, 2018-12-07:
Sorry, completely forgot to do this. I'll merge this and the fix that got added after 4.0.3 next week.

mseeley, 2018-12-10:
Hi @opdenkamp no worries + thanks a lot!

closed 2020-07-10 by opdenkamp