Pulse-Eight/libcec #443: Transmit call fails when size is > 12

srobins6, 2018-11-29:
In the LibCecSharp the Transit function fails when command.Parameters.Size > 12

malard, 2018-11-29:
maximum length for cec spec is this, we can put a guard to help developers

srobins6, 2018-11-29:
Where are you getting the maximum number? All the documentation I can find says it's 16, not 12.

malard, 2018-11-29:
the first 4 are taken by transmission data, @opdenkamp will confirm my thoughts

srobins6, 2018-11-29:
Does the "Initiator" value which I believe takes up 2 bytes at the start tell the adapter which of potentially multiple hdmi connections to use?

srobins6, 2018-11-29:
Can you elaborate on what specific information those 4 bytes represent?

opdenkamp, 2018-11-30:
Hi @srobins6 , I've responded to your question in the email to you've sent. You can send 16 bytes of data, 2 of which are taken by the header (initiator, destination, opcode). I've verified that this is working correctly on libCEC 4.0.2+

There was a bug in an old version of libCEC which prevented the last bytes from being sent, which has been fixed. If it's not working for you, then you may have to upgrade: https://github.com/Pulse-Eight/libcec/releases/tag/libcec-4.0.3

srobins6, 2018-12-02:
I am still having this issue with libcecsharp specifically, haven't tested with other libraries

opdenkamp, 2018-12-02:
I have tested sending a 16 byte packet with CecSharpTester: https://github.com/Pulse-Eight/cec-dotnet/tree/180cbe2b4e2b2d6a58c4b22d09760aba4b67a8e5/src/CecSharpTester
It was received correctly by another CEC adapter on which I was running cec-client.

Have you upgraded to the latest version?

opdenkamp, 2018-12-03:
Updating the status here too: this is caused by a bug in the firmware on the TX side, which results in TRANSMIT_FAILED_TIMEOUT_DATA when attempting to send out a 16 byte packet.

opdenkamp, 2020-04-27:
This isn't caused by the firmware after all, but was caused by a buffer that couldn't fit this.

commit 88894aa1 (2020-04-27): fixed: 15 and 16 byte frames didn't fit and couldn't be sent

closes #443
closed 2020-04-27 by opdenkamp