Pulse-Eight/libcec #444: Volume control not working after fix of #424

keydon, 2018-12-02:
Hallo,

I am experiencing a regression in the volume controls.
With the fix of #424 my volume controls within kodi are broken.
See me original bugreport in the kodi forum [here](https://forum.kodi.tv/showthread.php?tid=298462&pid=2795696#pid2795696).

Here are the kodi libcec enabled debug logs:
OK (before #424): http://ix.io/1uFj
Bad (after #424): http://ix.io/1uFn

I am using the Pulse-Eight USB-CEC-Adapter (latest firmware v7 2016-04-13) on regular x86_x64 hardware. And indeed, I have a Samsung-TV. But inbetween I also have a Denon AMP, which is supposed to control the volume.

opdenkamp, 2018-12-03:
Found the issue and will push the fix today. Thanks for the extensive report. Helps a lot to get bugs fixed quickly :)

keydon, 2018-12-03:
will do, thank you for the quick fix :D

keydon, 2018-12-03:
with the latest nightly milhouse build [#1203](https://forum.kodi.tv/showthread.php?tid=298462&pid=2797320#pid2797320)
I can happily confirm, the fix is working for me.
Thank you for the excellent bugfix-experience.

closed 2018-12-03 by keydon