Pulse-Eight/libcec #445: cec not working, HIGH_ERROR line:218

sbart76, 2018-12-21:
Hi Guys,

I have just installed a USB CEC adapter in my NUC, but I'm afraid it is not working. cec-client shows error as following:

```
kodi@kodi ~ $ cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.2
no serial port given. trying autodetect: 
 path:     /sys/devices/pci0000:00/0000:00:15.0/usb1/1-5
 com port: /dev/ttyACM0

opening a connection to the CEC adapter...
DEBUG:   [              23]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              24]	connection opened, clearing any previous input and waiting for active transmissions to end before starting
DEBUG:   [              24]	communication thread started
DEBUG:   [              35]	turning controlled mode on
NOTICE:  [              58]	connection opened
DEBUG:   [              58]	processor thread started
DEBUG:   [              58]	<< Broadcast (F) -> TV (0): POLL
TRAFFIC: [              58]	<< f0
DEBUG:   [              58]	setting the line timeout to 3
DEBUG:   [              95]	>> POLL sent
DEBUG:   [              95]	TV (0): device status changed into 'present'
DEBUG:   [              95]	<< requesting vendor ID of 'TV' (0)
TRAFFIC: [              95]	<< f0:8c
TRAFFIC: [             295]	>> 0f:87:00:e0:91
DEBUG:   [             295]	TV (0): vendor = LG (00e091)
DEBUG:   [             295]	expected response received (87: device vendor id)
DEBUG:   [             295]	replacing the command handler for device 'TV' (0)
DEBUG:   [             295]	TV (0): CEC version 1.3a
DEBUG:   [             295]	TV (0): menu language set to 'eng'
NOTICE:  [             295]	registering new CEC client - v4.0.2
DEBUG:   [             295]	>> TV (0) -> Broadcast (F): device vendor id (87)
DEBUG:   [             295]	detecting logical address for type 'recording device'
DEBUG:   [             295]	trying logical address 'Recorder 1'
DEBUG:   [             295]	<< Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [             295]	<< 11
DEBUG:   [             327]	CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [             327]	<< 11
DEBUG:   [             359]	CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG:   [             359]	>> POLL not sent
DEBUG:   [             359]	using logical address 'Recorder 1'
DEBUG:   [             359]	Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [             359]	Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [             359]	Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [             359]	Recorder 1 (1): CEC version 1.4
DEBUG:   [             359]	AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [             359]	setting ackmask to  2
DEBUG:   [             365]	Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [             365]	Recorder 1 (1): menu language set to 'eng'
DEBUG:   [             365]	GetPhysicalAddress - trying to get the physical address via ADL
DEBUG:   [             365]	GetPhysicalAddress - ADL returned physical address 0000
DEBUG:   [             365]	GetPhysicalAddress - trying to get the physical address via nvidia driver
DEBUG:   [             365]	GetPhysicalAddress - nvidia driver returned physical address 0000
DEBUG:   [             365]	GetPhysicalAddress - trying to get the physical address via drm files
DEBUG:   [             366]	GetPhysicalAddress - drm files returned physical address 2000
DEBUG:   [             366]	AutodetectPhysicalAddress - autodetected physical address '2000'
DEBUG:   [             366]	Recorder 1 (1): physical address changed from ffff to 2000
DEBUG:   [             366]	<< Recorder 1 (1) -> broadcast (F): physical address 2000
TRAFFIC: [             366]	<< 1f:84:20:00:01
DEBUG:   [             499]	using persisted autonomous mode setting: 'enabled'
DEBUG:   [             505]	using persisted CEC version setting: '1.4'
DEBUG:   [             510]	using persisted logical address setting: 'Recorder 1'
DEBUG:   [             515]	using persisted device type setting: 'recording device'
DEBUG:   [             521]	using persisted logical address mask setting: '206'
DEBUG:   [             537]	using persisted device name setting: 'CECTester'
DEBUG:   [             548]	using persisted physical address setting: '2000'
NOTICE:  [             548]	CEC client registered: libCEC version = 4.0.2, client version = 4.0.2, firmware version = 8, firmware build date: Fri Jul 21 09:08:24 2017 +0000, logical address(es) = Recorder 1 (1) , physical address: 2.0.0.0, git revision: <unknown>, compiled on Fri Dec 21 21:10:36 UTC 2018 by portage@hostname: Unknown host on Linux 4.15.12 (x86_64), features: P8_USB, DRM, P8_detect, randr
DEBUG:   [             548]	Recorder 1 (1): vendor = LG (00e091)
[... 147 lines cut ...]

```
21:58:47.522 T:139932125134016  NOTICE: Register - new keyboard device registered on application->keyboard: Keyboard (0000:0000)
21:58:47.533 T:139931292784384  NOTICE: Register - new cec device registered on cec->/dev/ttyACM0: CEC Adapter (2548:1002)
21:58:47.624 T:139932125134016  NOTICE: Register - new mouse device registered on application->mouse: Mouse (0000:0000)
```

yet it still is not working with the remote.

Neither anything suspicious in the dmesg:
```
[    1.204267] usb 1-5: new full-speed USB device number 4 using xhci_hcd
[    1.336701] usb 1-5: New USB device found, idVendor=2548, idProduct=1002
[    1.338794] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.340852] usb 1-5: Product: CEC Adapter
[    1.342920] usb 1-5: Manufacturer: Pulse-Eight
[    1.344990] usb 1-5: SerialNumber: v8
[    1.347799] cdc_acm 1-5:1.0: ttyACM0: USB ACM device
[    1.351402] input: Pulse-Eight CEC Adapter as /devices/pci0000:00/0000:00:15.0/usb1/1-5/1-5:1.2/0003:2548:1002.0004/input/input19
[    1.353827] hid-generic 0003:2548:1002.0004: input,hidraw3: USB HID v1.10 Mouse [Pulse-Eight CEC Adapter] on usb-0000:00:15.0-5/input2
```

Same TV worked well with CEC functionality from Raspberry Pi. I would apppreciate a lot any help.

Bartek

sbart76, 2018-12-29:
I read a response from other thread, and switched off the CEC setting in BIOS. Everything works perfectly now :)

closed 2020-07-10 by opdenkamp

LeeRuns, 2022-01-23:
@sbart76 how did you resolve this? Please link the other thread.