Pulse-Eight/libcec #450: Python Binding - Unbuntu not working

Lcpuche, 2019-01-29:
Hello,

I followed all the steps and got it working in a RASPBERRY PI with no issues. However, trying to follow the Ubuntu instructions yields this error:

```
  File "/media/sf_shared/Transferer/pyCecClient.py", line 39, in pyCecClient
    cecconfig = cec.libcec_configuration()
AttributeError: module 'cec' has no attribute 'libcec_configuration'
```

Anyone know how to resolve this?

neilmunday, 2019-04-29:
I have the same issue under Arch Linux on the Raspberry Pi using Python 2.7.

neilmunday, 2019-04-30:
If it helps, it works fine if you checkout tag libcec-4.0.0.

neilmunday, 2019-04-30:
Just to add, under Fedora 27 (x86_64) the Python bindings work fine with Python 3 but not Python 2.7.

Steps to reproduce:

```
git clone https://github.com/Pulse-Eight/libcec
cd libcec
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/home/neil/test -DPYTHON_INCLUDE_DIR=/usr/include/python2.7 -DPYTHON_LIBRARY=/usr/lib64/libpython2.7.so ..
make -j
make install
```
Now try importing and using the module:

```
export PYTHONPATH=/home/neil/test/lib/python2.7/site-packages
$ python2
Python 2.7.15 (default, Oct 15 2018, 18:36:25) 
[GCC 7.3.1 20180712 (Red Hat 7.3.1-6)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import cec
>>> cec.libcec_configuration()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'libcec_configuration'
>>>
```

opdenkamp, 2020-07-10:
closing this one because it's been fixed already

closed 2020-07-10 by opdenkamp