Pulse-Eight/libcec #451: Physical Address allocation in CecSharp : Recording Device 1(Pulse eight) and Audio System( Sound bar) allocated same Physical Address.

vishwanath-tns, 2019-02-06:
Hi,
      I am using Pulse-eight Cec Adapter on windows platform. Cecsharp.dll is used to scan the devices. When scan is executed, both Recording Device 1, and Audio System are allocated same physical address. 
Please help! Following is the output

[image]

Regards
Vishwanath

vishwanath-tns, 2019-02-06:
Cec Library Version as mentioned in below screenshot

[image]

opdenkamp, 2020-07-10:
you have to set the `BaseDevice` and `HDMIPort` properties to the correct value when the address can't be detected by reading the EDID: https://github.com/Pulse-Eight/libcec/blob/master/src/dotnetlib/CecSharpTypes.h#L1773

try upgrading to the latest version first. if it still isn't detected correctly, set the properties mentioned above

closed 2020-07-10 by opdenkamp