Pulse-Eight/libcec #454: AVR changes input when TV is switched on

biryets, 2019-03-10:
**Issue:**

When the TV is turned on, after about 10 seconds, the AVR automatically switches its input from the Raspberry Pi to its own home screen, without any user input. The Raspberry Pi and Sony AVR is left on at all times, so this is not related to the Pi turning on. 

**To reproduce:**

1. Disable CEC on/standby synchronisation on TV, AVR and  RPi
2. Switch TV to AVR  and AVR to RPI input
3. Stop  audio or video playback on RPI 
4. Power off TV leaving AVR and RPi on
5. After 5 mins bring TV out of standby using TV remote 
6. TV will turn on, AVR will initially output RPi and after 5-10 seconds AVR will automatically switch from the RPi HDMI input to its own hone screen. 

**Hardware:**

- TV - LG OLED65B7
- AVR – Sony STR-DN1080 
- Raspberry Pi 3 , Veero 4k,  Minix U1
- HDMI connection between Pi →AVR→ TV 

**Software:**

- RPi running Libreelec 9.01,  OSMC  2018.12-1, Coreelec 9.0.1.
- It has been an issue for at least the last 3 months. Possibley longer
- The issue appears linked to the updated version of libcec that is bundled with Kodi Leia. 

**Thoughts:**

Appears to have similar symptoms with the following issues:
https://github.com/Pulse-Eight/libcec/pull/256
https://github.com/Pulse-Eight/libcec/pull/372
https://github.com/Pulse-Eight/libcec/issues/307

It seems like either the Pi is sending some kind of signal it shouldn't be, or the AVR is incorrectly picking up something the Pi is sending and changing input when it's not supposed to.

biryets, 2019-03-21:
I've patched my system to reverse the effects of this commit https://github.com/Pulse-Eight/libcec/commit/9cc4c29aada468eed976b70966035ca933029c1c#diff-5e82a8a73dfb85574a98f4b64dfcb842 and now my system is working fine. This may not be the solution for everyone but it works for me.

bummzulu, 2019-09-08:
I have an issue that seems to be similar. 

I am using LibreELEC with the latest LibCEC Version 4.0.4.

My setup and issue description:

Raspberry Pi with LibreELEC and LibCEC in V4.0.4 and connected to TV (or AVR directly - tried both with the same result)
AVR Denon AVR-X2000

When the AVR is turned on and playing music from an anlog source the Raspberry Pi seems to send an unwanted signal and the AVR switched to the HDMI input.

As I am on the latest version I am afraid that the fix from issue 307 does not fix the issue in my scenario.

elfhack, 2020-10-12:
Same here, and this issue has been plaguing me for some time with different hardware
- Samsung UE48H6700 (Old 2015ish LED TV 1080p)
- Currrent LG OLED55B9 
- Yama RX-V679
- NUC 6i5SYH with Pule-Eight embedded addon (running kodi)
- USB Pulse-eight CEC module (running kodi)
- ODROID C4 (running coreelec - which is kodi)

And it's ALWAYS the same behaviour. A source turns on the TV (be it a PS4, Switch or just the remote), the TV turns on the AVR with the correct source, a couple of seconds pass and the device running kodi with libcec steals the show and switches the AVR to itself. It's beyond infuriating, and what's worse is as far as i know i can't sniff the exact time this happens since kodi takes control of the CEC device, so I'd need a second one connected to the bus

ishioni, 2021-01-08:
Same here - Yamaha RX-V679. Multiple TVs - Samsung, LG, Sony, multiple client devices - USB CEC adapter, NUC adapter, Raspberry PI 3B, ODROID C4, all running some form of Kodi. This has plagued me many years and is infuriating

sookoll, 2022-10-28:
Same here. Raspberry PI 2, LG TV, always when switch on TV, after 5 seconds osmc took over. It starts after last Kodi update.

closed 2025-02-27 by github-actions[bot]