Pulse-Eight/libcec #456: Is it possible to launch an app on TV using HDMI cec command?

vijayvikma, 2019-05-02:
Hello Experts,

I have been trying and using cec-client commands referring to this website http://www.cec-o-matic.com/ .

I am not able to find/recognize anything which can launch an app(e.x. YouTube).

I am using raspbian OS on pi 3b+ and having connected to my android TV on HDMI port 1.

malard, 2019-05-02:
No :)

From my phone

vijayvikma, 2019-05-02:
@malard Thanks for the clarification.
Somehow I managed to get a workaround by executing up/down/select key command.:)
Below are the step/flow  working for my tv.
pi@raspberrypi:~ $ cat openyoutube.sh 
echo 'tx E0:44:09' | cec-client -s -m;#go to tv home menu

echo 'tx E0:44:02' | cec-client -s -m;# down
echo 'tx E0:44:02' | cec-client -s -m;
echo 'tx E0:44:02' | cec-client -s -m;

echo 'tx E0:44:04' | cec-client -s -m;# right
echo 'tx E0:44:04' | cec-client -s -m;
echo 'tx E0:44:04' | cec-client -s -m;

echo 'tx E0:44:00' | cec-client -s -m;# select

echo 'tx E0:44:02' | cec-client -s -m;
echo 'tx E0:44:04' | cec-client -s -m;
echo 'tx E0:44:00' | cec-client -s -m;

closed 2025-02-27 by github-actions[bot]