Pulse-Eight/libcec #458: Any way to disable CEC or have it enabled solely on a given application?

oblitum, 2019-05-21:
Hi, my Pulse-Eight adapter just arrived and it's working like a charm in my Linux PC. I have `libcec` installed. I'm using it solely when I launch kodi (it's not a kodi-only machine), but regardless of that, having the adapter connected causes makes my TV controller useless for standard TV settings, I can't control brightness, etc, because the adapter has the CEC focus over my TV, which is only useful when I'm using kodi.

So, is there any command I can do to at least disable CEC for my PC and let the TV take control for a while?

Another question is whether it's possible for my PC have CEC enable only when I start kodi.

oblitum, 2019-05-21:
I've noticed that while CEC restarts when I do `echo r | cec-client -s -d 1`, sometimes I'm able to get standard TV controller functions back, but just meanwhile it's restarting. I have no idea of other commands for  my purpose.

malard, 2019-05-21:
We sell a CEC-Less HDMI cable that you can fit between the adapter and the PC, but without knowing what PC you have, its impossible to say how, but probably it will be in your BIOS

oblitum, 2019-05-21:
I'm not sure what's the use of that cable in this case because I don't want to disable CEC permanently. I bought the Pulse-Eight adapter to control Kody in my PC, and it works, but the side effect is that it permanently takes control and I can't use my TV controller to adjust brightness for my PC. The TV is my monitor.

oblitum, 2019-05-21:
That's the reason why I'm asking whether I can do anything to have CEC only when I start kodi, or some command to stop/start CEC manually so I can use my TV remote.

malard, 2019-05-21:
You would have to seek that in Kodi, not in our library

oblitum, 2019-05-21:
Sorry, it doesn't make sense. kodi is the application where having CEC is useful, but it could be any other. My problem is not something that gets triggered when I start kodi, on the contrary. Due to the adapter being connect between my PC and TV, I lose standard TV remote functionality, always, even when I'm not running kodi. The simplest example is that I wish to control brightness, but I can't anymore unless I detach the adapter from USB.

oblitum, 2019-05-23:
I was able to configure CEC behavior in my TV settings and changing the CEC keys from "Menu" to "Normal". This allocates fewer keys in the TV remote for CEC control, but enough to basic Kodi navigation. I can now control my TV brightness, color, etc again by clicking the "option" key, sadly now I lose the context menu functionality inside Kodi, since the context menu is triggered by the "option" key, which isn't allocated for CEC anymore.

I'm closing the issue, despite not obtaining the answer I was looking for (temporarily disconnect/turn-off the adapter or having it enabled solely when I start Kodi).

closed 2019-05-23 by oblitum