Pulse-Eight/libcec #459: Samsung TV turning on by itself

TeHashX, 2019-07-02:
Hi, I have a Samsung TV 55MU7002, intel Nuc NUC8I7BEH with usb - cec adapter, latest cec-firmware-v9.zip but tv it still powering on by itself sometimes, how can I completely disable on/off from windows 10?

kdschlosser, 2019-07-13:
OK I need you to verify the problem is stemming from the CEC adapter. 

There is a known issue with Samsung TV's that power themselves on at random times. There have been many attempts to identify the cause. This issue has plagued Samsung TV's for several years now and Samsung simply ignores there is a problem. 

Some have stated this issues is from using HDMI-ARC (Audio Return Channel) 
Others have said it is from a linksys or netgear router sending some kind of an odd packet
There have been people that have stated it is from CEC..
There have people that stated it is a sound bar
There are also reports of the TV doing this because it is checking for updates (TV has to be on for the network adapters to function properly)
There are also reports of CEC being turned off. the automatic updates being turned off not connected to a network and HDMI-ARC not being used/turned off. No inputs of any kind plugged in and the TV still turns it's self on. 

None of these claims provide any information on how they came to the reason. and non of them state what the fix is.

So you will need to unplug the CEC adapter and wait and see if the TV stops turning on. If it does stop. try plugging into a different port on the TV. (make sure you set the port correctly in whatever application or code you are using). then see if it still occurs. if it stops take note of the port number and also if that port supports HDMI-ARC and provide the information here. You may find yourself chasing a ghost

The end game is this.. You have the solution in your hands no matter what is causing it. The CEC adapter reports remote button presses (and this works for Samsung TV's). If a button press is not received for the TV to power on. a program can easily be written to turn the thing back off if it turn on. The only issue is that you would not be able to use the power button on the TV it's self. But this i feel is a small compromise to have to make (considering most people do not even use it) in order to not have the thing turn on at 2AM in the morning and scare the snot outta ya.

thewarm, 2019-07-18:
I have a NUC8i3 and a Samsung 65MU7500. I discovered that if I disable "Anynet+" the TV will NOT wake up. I also lose all CEC control. But, if I enable "Anynet+" (no other changes) the set WILL wake up whenever it feels like it.
I am running libCEC-4.0.3 with the V9 firmware.
Hope this helps fix the problem...

kdschlosser, 2019-07-19:
OK lets dig a little deeper.

There are several more questions so we can rule out what you have and have not tried. We want to isolate the actual cause of the problem.

If you do not have the CEC adapter plugged in and ANYNET+ turned on does it do it?

Do you have a soundbar attached? if so have you tried unplugging that and checking if it still happens? There have been reports of soundbars causing the problem. 

Have you tried unplugging all HDMI devices Enabling ANYNET+ does the problem occur. If this one is a yes it does then there is not a whole hell of a lot we can do except possibly band-aide the problem and use CEC to turn the TV back off.

I am thinking of the number of times I have seen people mention that the problem is ANYNET+ (HDMI-CEC), Audio Return (HDMI-ARC) and soundbars. I am beginning to wonder if the issue is because of HDMI-CEC and HDMI-ARC the only thing that typically  uses both of these things are soundbars. There could be an issue internally in the TV's if CEC and ARC are being used at the same time. If you do not have a soundbar and or you are not using HDMI-ARC then this would not be the case. 

I know this is a big pain because of how inconsistent the TV turning on is. If you are willing to walk through the steps with me hopefully we can isolate the cause and create some kind of a solution.

thewarm, 2019-07-19:
I do not use a soundbar or AV receiver. I only have my NUC8i3 plugged into
HDMI 1 (tried the other inputs, same issue). I have not enabled Anynet+
without the CEC board (why would I?).

Here is something strange that just happened... I installed a SiliconDust
HDHomeRun network tuner (that I use in Kodi 18.3). I unplugged my TV
antenna and removed "Live TV" from the "Smart Hub". The TV did NOT turn on
(all day). I will see if it stays off overnight.. I had already unchecked
auto-updates to the firmware and the apps, but maybe the TV was still
"phoning home" looking for an OTA update??

Thank you, Steve

I have the latest TV firmware. I have the latest BIOS for the NUC (with CEC
unchecked in BIOS)

kdschlosser, 2019-07-19:
It could be an application doing it?? IDK it's a complete mystery. I do know the exact same problem you are having occurs for people without the use of libCEC. So I can say at almost 100% certainty that libCEC is not the issue.

The funny thing is that no one has managed to pinpoint an exact cause, not even Samsung because this issue has been plaguing there TV's for about 5 years now with no solution. and no known cause. 

Samsung does what Samsung does when they have no clue what is causing it, or there is a whole mess of warranty claims. They simply ignore the complaints then "discontinue" the parts and tell you 

> Sorry but there are no parts available for your TV because the manufacturer has discontinued them, so we will not be able to repair it under warranty. for the inconvenience here is a $100.00 off coupon towards another Samsung TV

that they will then discontinue the parts for as well. 

What they do not tell you is that "Samsung" discontinued the parts in their system. The manufacturer actually did not and with some exhaustive searching you can locate the actual manufacturer and find out that they are in fact still making the "discontinued" parts. 

The above is from experience, and the reason as to why I will never purchase another Samsung product. 3000.00 USD on a washing machine and dryer and in less then 4 years the washer breaks and I come to find out that the parts have been "discontinued" and the breakage was on the drive system which carried a 10 year warranty and is a common problem with Samsung washing machines. 

You should read up on the antics that Samsung does.

> I have not enabled Anynet+ without the CEC board (why would I?).

at this point is would be to isolate the cause of the problem. to see if it is in fact libCEC or the TV. but to complete this kind of a test you would need to have access to another device that you can plug into the TV that has CEC available on it, like a blue-ray player. Because if you unplug the CEC adapter and the problem goes away does not mean that the CEC adapter is the cause. if you then plug in that other device and the problem comes back. then the TV is to blame. The rule of the common denominator applies here..

You can also try changing out your HDMI cable. There is data that rides up and down these cables. CEC is one of such types of data. and as it so happens CEC has the ability to turn on the TV.. Some data could be getting corrupted in transit through the cable. and the TV is interpreting the corrupted data as "turn on"

I am trying to think of all of the things you can try in order to solve the issue.

You mentioned the Antenna connection and disconnecting it. Did you have the coax cable for the antenna bundled together with the HDMI cable? possibly zip tied together? or maybe in the same cable chase in close proximity?

You can also have a look at this repository you can use it in conjunction with CEC or as a CEC alternative if you are still having issues with CEC.

https://github.com/kdschlosser/samsungctl

 I am planning on adding that band-aide i mentioned into the project  I am still working on it, took a break for a bit and I am about to jump back into working on the thing again. That program offers the most control available for Samsung TV's. It even offers more control then Samsung's own applications. It supports TV's from 2011 to current.

thewarm, 2019-07-19:
I read through the Samsung forum (too many pages) about this issue and also
the auto-switching to the active HDMI port (which I was able to disable
with a code via the remote).
BTW, the TV was on when I got up this morning. I put my NUC in S3 sleep
before I turned off the TV last night. The TV woke up the PC. I have tried
multiple HDMI cables to no avail. The only thing that I know *will *work is
turning off Anynet+ before I turn off the TV.
Makes me want to pull an old Harmony remote out and use IR. I do not have
this problem with my Sony XBR.

I just had to get a curved TV when I retired...

Thank you, Steve

thewarm, 2019-07-19:
I'm throwing in the towel. I turned off my NUC's internal USB headers (I
left the CEC board inside). I switched to IR control. Not ideal, but it
works. I do miss the Samsung's tiny remote!
Let me know if I can help you test ideas...
Steve

thewarm, 2019-07-19:
I like the little Samsung remote too much to give up.. I am going to try an
experiment. I set my onboard USB headers to "not detected" in the NUC's
BIOS. Tonight when I go to bed I will power off (S5) the NUC. Not detected
means that the CEC board will not be recognized by BIOS, but only by the OS
(Windows 10 Pro).
I will let you know tomorrow if this made a difference. If it does it will
only help NUC owners..
Steve
frustrated in Seattle

kdschlosser, 2019-07-20:
Trust me I feel your pain. Samsung has gone into the crapper. batteries that explode, TV's that turn on whenever they like. Washing machines that literally self destruct and throw parts out of them injuring people...

I have an idea on how to band-aide this issue. It will not be a proper fix as only Samsung is capable of doing that. But using CEC I should be able to turn the TV off if it turns on by it's self.

There are a couple of things I do not know. I know that libCEC does receive notifications of remote button presses from the OEM remote. I do not know if the power button is one of these buttons.

If it is... this is what I am thinking. we have a flag that gets set to True if either a CEC power on command takes place or a notification for the power button being pressed on the remote happens. then when libCEC gets the notification that the TV has powered on it will then check that flag. if the flag is True then everything goes about it's business. If the flag is False then libCEC will turn the TV off. False will indicate that no CEC command turned the TV on and the power button on the remote was not pressed. The only issue with this approach is the power button directly on the TV, we have no way of knowing if that button gets pressed at all. The button will work fine for turning the TV off. but if you use it to turn the TV on the program will turn the TV back off again. 

The other thing is the NUC will have to remain fully powered on (no sleeping) in order for this to work.

If this is acceptable for you I can key something out pretty quick. It will be a test program and it will require that you have python installed. I will have to create an installer for it but that is not to much of a big deal.

You also stated you are running Windows 10...

You may want to give this little application a try.
http://www.eventghost.net
be sure to download version 0.5

Here is a plugin for the CEC adapter for that application
http://www.eventghost.net/forum/viewtopic.php?f=9&t=10018

and you also mentioned kodi.
http://www.eventghost.net/forum/viewtopic.php?f=9&t=1562

If you use that application you will not have CEC handled by kodi. instead you have it handled by that application and then use that application to interface with kodi and hand off the remote button presses to kodi. 

I am the administrator and one of the authors of that application, I also wrote the CEC interface for it as well. If there is something you need help with send me a message on the forum and I will be happy to help out in any way that I can.

thewarm, 2019-07-20:
During my "frustration period" I uninstalled libCEC. I only have CEC
control when in Kodi (which is what I want). For last nights experiment, I
powered my NUC completely off (S5). The Samsung woke my NUC!
I woke up and turned of Anynet+. It's 3AM..

kdschlosser, 2019-07-20:
I believe that Kodi uses libCEC so it may not be uninstalled.

If you give that application a try I will be able to walk you through some tests really easily. it will provide us a mechanism to view what is going on with the CEC network before during and after the TV powers on by it's self. It could help us in determining what i happening.

thewarm, 2019-07-21:
Kevin
I found a solution. I enabled IR control on my NUC so I could use a Harmony
remote. I ended up not using the Harmony. But this morning I went to the
'Universal Remote' and tried 'Game Console\X-Box' and it works! I can use
the Samsung remote without CEC.
I am going to stop right there.
It is too hot to play with Anynet+ at this time... maybe when it cools off

Thank You

TeHashX, 2019-09-20:
There is a chance to release a special firmware update version with poweron / power off function completely disabled?
It become a nightmare, tv turning on by itself 3-4 times at night.
Thank you

malard, 2019-09-20:
Its not the firmware of the adapter doing it, you can modify libCEC to disable this function

TeHashX, 2019-09-20:
Which file to modify on Windows 10?

thewarm, 2019-09-20:
I gave up on Anynet. I now use IR control on my NUC8i3 and configured it as
a X-Box Game Console.

TeHashX, 2019-09-21:
I dont't want to use 2 remote controls and I'm sure power on/off functions can be disabled in firmware.

kdschlosser, 2019-09-22:
The funny thing is the problem is not with the CEC adapter or the firmware.. there are reports of this same thing happening with people that have never even heard of the HDMI CEC adapter.. The problem is the TV. this problem has existed now for about 4 years. and Samsung's response to it.... no response at all. they are ignoring the problem hoping it is going to go away.... The reason is it's crappy cheaply built products.. some have the issues while others do not. if they acknowledged the problem they would be replacing a whole lot of TV's 

The best solution is to NEVER purchase another Samsung product again.. They make a product they do not stand behind. and they build it as cheap as possible. And the help Samsung gives is their phone support which is an absolute JOKE the people have not a clue and their job is to war you down so you give up. the same thing applies with their warranty.

I will almost bet if you went and purchased a new mainboard for the TV the problem will go away. most Samsung mainboards are in the 100 to 200 area (USD) it could be more or it could be less. it depends on how new/old the TV is. There are a plethora of videos and walk throughs on how to disassemble the TV's. This is the only suggestion I am able to give.. The problem is the TV and not anything you are plugging into it. Fix the problem instead of band-aiding it. because we all know that band-aides fall off eventually.

Nekmo, 2020-03-08:
I have a Samsung TV and I have never had this issue with my Nintendo Switch, which uses HDMI CEC.

However, my TV turns on randomly every few hours with a pulse-eight adapter in my HTPC.

I don't use sound bars or anything like that. I would like to find a solution to the issue, either by debugging the driver or deactivating it when the TV is not in use.

thewarm, 2020-03-08:
Issue is with the Samsung. I gave up on CEC control with this TV. I now use
IR.
BTW, my TV liked coming on in the AM by itself too with the Pulse Eight in
an Intel 8th Gen NUC (only on the Samsung, works correctly on a Sony XBR
and a 10 year old Vizio!)
Sorry, I have no suggestions for you.

opdenkamp, 2020-03-28:
Sorry for the delay. This is a bug in Samsung's 2016+ TVs. A new firmware for the CEC adapter that works around this bug is on it's way. Testing it now and I expect to have an update ready early next week.

kdschlosser, 2020-03-28:
Awesome..

I do have a question regarding this. People have brought up this issue even without using a pulse eight adapter and they only have a soundbar plugged in. This would make it a poblem in the TV it's self yes? another thing is that some TV's have the problem and others do not, even when they are running the same firmware on the TV's. so that removes the issue from being a software issue and it ends up being a component problem correct?

There have been reports of the problem possibly being in the ARC (Audio Return Channel) Others have stated the issues is with CEC, There are even reports of it being caused by a router on a network. Unfortunately peoples troubleshooting skills lack the precision needed to identify the actual cause and this makes it hard to nail down the exact cause. 

I thought that the issue might be able to be "band-aided" using CEC.

If it can get fixed that would be fantastic.

opdenkamp, 2020-04-02:
can you try this (beta) firmware please: [cec-firmware-v10.zip](https://github.com/Pulse-Eight/libcec/files/4420698/cec-firmware-v10.zip)

@kdschlosser correct, the issue can be caused by _any_ CEC device on the bus. it's caused by random CEC traffic on the bus and it's causing 2016+ Samsungs to power on randomly. I think that it's some buffer overflow in the TV's CEC implementation that's causing it to crash and restart or to somehow trigger the power on command. I can only prevent our own devices from doing any CEC while the TV is off and I can't prevent traffic from other devices from reaching the TV. Only the TV vendor can fix this.

Some Android TV Sony TVs have a similar issue, returning the wrong power state while the TV is off sometimes, so probing is disabled for Sony too in this firmware while the TV is off.

It also adds a new feature that many have requested: send a power on command to the TV when the PC is powered on. That feature is disabled by default (to not change behaviour) and can be enabled using the latest libCEC/cec-client from Github and the upcoming 5.0 release: `cec-client -aw 1`

opdenkamp, 2020-04-02:
oh and when using this firmware and older versions of libCEC: ignore warnings about HDMI_CEC_VERSION commands being rejected. that command had to be dropped to free some rom space and isn't required.

TeHashX, 2020-04-02:
I get this error while trying to update

[image]

opdenkamp, 2020-04-02:
please download and run https://www.microsoft.com/en-us/download/details.aspx?id=5555 and then run the upgrade again

TeHashX, 2020-04-02:
It worked, now wait and see if the tv will randomly power on by itself again.
Thanks
[image]

kdschlosser, 2020-04-04:
@opdenkamp 

While this does stop anyone using a CEC adapter from encountering this issue because of the CEC adapter the problem still exists So people that have other CEC devices can still have this problem.

Now libcec is able to get the remote button presses from the OE remote control. My question now is.. Does it get the power button being pressed??. I ask this because if it does and the TV is in a standby state then it would be simple enough for libcec to turn the TV off if it powered on and there was no cec command issued for the TV to power on and the remote button was not pressed. I doubt this is a firmware issue with Samsung TV's, I say this because there are people with the same hardware where some have the problem and others do not. running the same firmware version. If this was a firmware problem in 4 years time Samsung would have fixed it. I believe this is a hardware issue and it's a hit or miss problem. Samsung being the whore of a company that it is has deemed that the increased cost to replace the flaky component with one that is not going to have an issue is going to take to much money from them. And that there are not enough TV's encountering the problem compared to how many TV's they sell. So they just ignore the people having the issue. Firmware is easily updated. 4 years is a long time to not fix this kind of a problem. But Samsung is also an ass backwards company and the departments work independent of one another and do not inform each other of changes that are made. 

Now you stating that Sony TV's also have the same issue.... What do you think the odds are that 2 different companies made the exact same programming errors to creat the exact same issue in thh firmware??.. Now what are the chances that 2 different companies buy the same components used inside of their TV's.. This problem HAS to be a hardware problem and not a software one.

It would imagine it would be worth looking into. If it can be done that would mean additional revenue for PulseEight because your device would be the only thing available on the market that would provide a workaround for the problem. I do not know what information is available from a TV. is there a way to log all CEC data no matter what it is. Just dumping all of the data into a file as hex?. It would make reverse engineering of OE additions or changed to the protocol easier to do without having to own the device.

I also wanted to ask how you guys are holding up over there in the UK??. I wish all of you the very best of luck in the coming months. This is one hell of a thing taking place.

opdenkamp, 2020-04-04:
> While this does stop anyone using a CEC adapter from encountering this issue because of the CEC adapter the problem still exists So people that have other CEC devices can still have this problem.

Correct, and that's exactly what's happening depending on what the source devices that are connected to the TV are doing.

> I doubt this is a firmware issue with Samsung TV's, I say this because there are people with the same hardware where some have the problem and others do not. running the same firmware version. If [...]

I've first encountered this problem on Pulse-Eight's matrix switches, when some customers started to complain that had 2016 Samsungs. I've debugged the issue using that hardware, which allowed me to investigate in detail and I can assure you that nothing is happening on HDMI that should cause the TV to power on. It's random CEC traffic, even CEC polls, and at some point the TV powers on and sends the same CEC frames that it sends when powered on with the remote. The TV stopped doing this after I prevented the matrix from doing CEC while the TV is off. I assume that it's some buffer overflow or other crash in Samsung's firmware, but I didn't investigate further after my problem was fixed.

Samsung's CEC implementation changed completely in 2016 and it also introduced some other issues that I had to work around on the matrix switches. This specific issue probably hasn't been fixed by Samsung because it only happens when a device is connected that sends out CEC frames while the TV is off, and most devices don't do this. So it only affects a low percentage of their customers.

> Now you stating that Sony TV's also have the same issue.... What do you think the odds are that 2 different companies made the exact same programming errors to creat the exact same issue in thh [...]

Sony's issue is completely different. It won't power on randomly like Samsung, but some AndroidTV Sony models sometimes return the wrong response to the request power state command (returning on instead of standby once).

Both are TV firmware issues that only the manufacturer can address.

> Now libcec is able to get the remote button presses from the OE remote control. My question now is.. Does it get the power button being pressed??. I ask this because if it does and the TV is in a [...]

I'm not sure what you mean, but see the explanation above about what's happening exactly. libCEC also doesn't know the difference between the TV powering on randomly like this and someone pressing the power button of the TV's remote.

> It would imagine it would be worth looking into. If it can be done that would mean additional revenue for PulseEight because your device would be the only thing available on the market that would [...]

cec-client will dump the traffic as hex to your console, but you can't use it while Kodi is using the adapter. And you can work around this problem in hardware already by using a Pulse-Eight video matrix now because it separates all the CEC lines and I'm in full control of what's sent to the TV there.

> I also wanted to ask how you guys are holding up over there in the UK??. I wish all of you the very best of luck in the coming months. This is one hell of a thing taking place.

The same as everywhere else right now I assume, but I live in the Netherlands myself ;-)

kdschlosser, 2020-04-05:
OK let me try to word this differently.

In libcec a callback can be registered for when buttons get pressed on the remote. 
Will this include the power button on the remote?
Will it report the power button being pressed if the TV is in standby?

If the above answers are Yes then that mechanism can be used to know if a person is wanting to turn the TV on.

If the TV is known to be powered off and no communications have taken place for the power button on the remote being pressed. And there has been no CEC command issued from any other device to turn the TV on (I am assuming the communications between say a BD Player and a TV can be seen by the P8 adapter), then a reasonable assumption can be made that the TV turned on by it's self. 

At that point the CEC adapter can send the command to turn the TV off. This is not something that would get used all of the time. It seems like it happens at completely random times. It is not a set time table that I am aware of. Most complaints are that it happens every few days usually in the middle of the night.

I know that this does not cover the power button on the TV it's self. But in all reality how often does someone even use that button. This would be an option that can be turned on and it would include a warning about the power button on the TV not being able to be used to power the TV on.

 I know it is not the proper fix and would only be a band-aide if it can even be done. After 4 years I highly doubt that Samsung is going to fix the problem. So something would be better then nothing.

TeHashX, 2020-04-05:
My tv turned on 4 times last night, and was on hdmi 1 input, with firmware v9 was on tv input or where did I leave it before turned off.
I will go back to v9 if possible.

opdenkamp, 2020-04-06:
@kdschlosser unfortunately that's not how it works. There is no power button press that can be ignored and you can't tell the difference between the tv powering on because of someone pressing the button on the remote and the tv powering on when it shouldn't. As for fixes: it affects all 2016+ Samsungs, so if it gets fixed then it might be fixed for all models.

@TeHashX I've found out the problem and I'm looking into it now. I hope to have another update later today. You can download the latest non-beta version here if you want to revert https://www.pulse-eight.com/Download/Get/56

opdenkamp, 2020-04-06:
@TeHashX please try [cec-firmware-v11.zip](https://github.com/Pulse-Eight/libcec/files/4438587/cec-firmware-v11.zip)

opdenkamp, 2020-04-06:
@TeHashX please try [cec-firmware-v11.zip](https://github.com/Pulse-Eight/libcec/files/4438616/cec-firmware-v11.zip) (wrong file uploaded a couple of minutes ago. this is the correct one)

TeHashX, 2020-04-06:
I can't install, see video recording https://streamable.com/ucc08b

TeHashX, 2020-04-06:
Ok, tried the second uploaded file and installed successfully, I will give feedback tomorow, thank you.

Nekmo, 2020-04-06:
@opdenkamp I am testing the latest release of libcec (branch master) on Arch Linux.

Commit: 8b0393ad27cf4f1bf51c228bdf82365ea9bfe90a
TV Model: Samsung TV MU6100 (2017)
Package: https://aur.archlinux.org/packages/libcec-git/
Linux Kernel: Linux 5.5.9-arch1-2
Pulse-eight device: 2548:1002 Pulse-Eight CEC Adapter (date of purchase: 2014)

I will be testing it for the next few days :+1:  Thanks!

opdenkamp, 2020-04-06:
@TeHashX thanks for testing

@Nekmo if you're testing the Samsung wake up issue: the firmware upgrade to v11, which should fix the problem, is only available for Windows. You only have to run it once and you can use any Windows machine to do the upgrade. Thanks

Nekmo, 2020-04-06:
@opdenkamp I don't have Windows available, sorry :(

TeHashX, 2020-04-07:
> @opdenkamp I don't have Windows available, sorry :(

Can you try running a vmware windows image and update firmware? 

My TV didn't woke up last night with firmware v11 :) 
Let's wait a few more days...

opdenkamp, 2020-04-07:
> Can you try running a vmware windows image and update firmware?

That won't work without jumping through hoops, because the USB device id and vendor id change when the device jumps into the bootloader, to get it to upgrade.
I'm looking into a statically compiled version for Linux, but that code still needs some work, so no ETA for that one.

> My TV didn't woke up last night with firmware v11 :)
> Let's wait a few more days...

Great, thanks for testing. The previous version was still sending CEC polls to the TV, to check whether it's still responding, and that caused the TV to bug and power on again. I've tested it on my own setup for almost 24h now and the CEC line has been silent and the TV hasn't powered on, so I'm closing this ticket now. Feel free to re-open it should it happen again, but unless USB power gets reset, there won't be any CEC activity initiated by the adapter anymore.

closed 2020-04-07 by opdenkamp

Nekmo, 2020-04-10:
@opdenkamp  I have used an old laptop to install Windows 10. For some reason the laptop's HDMI did not work but recognized the adapter using only the USB connector.

I have upgraded the adapter to version 11 and it seems to have worked. At the moment the television has not been turned on. :)

TeHashX, 2020-06-21:
I found the tv turned on just two times in 2 months, it's not perfect but a huge improvement with latest firmware.