Pulse-Eight/libcec #460: Turn off TV without turning off receiver.

ned-kelly, 2019-07-02:
I'm currently using `cec-client` to power off the TV as part of my home automation by sending `echo 'standby 0'` -- Sending this command however turns of the Receiver which is also connected to the TV (In my case, the media devices are actually connected to the Input of the Receiver, and then the HDMI output goes to the TV).

Is it possible to turn off ONLY the TV without turning off the receiver by using the custom `tx` commands and if so what should I be looking at? -- I'm currently using `tx 5f:82:31:00` to change the input on my receiver (device 5) however I can't work out how to turn on/off a device connected to the receiver.

malard, 2019-07-02:
Don't broadcast the off target the TV instead

40:

Instead of

4F: for example

From my phone

ned-kelly, 2019-07-02:
> Don't broadcast the off target the TV instead 40:

Sorry for my noob questions here - Still trying to get my head around the TX format --- So I thought the TV was always device 0? -- Shouldn't it be something like `tx 10:36` (Just read that 36 is a command to turn on/off the TV elsewhere)

closed 2019-07-17 by ned-kelly