Pulse-Eight/libcec #462: PC does not wake on TV power on (anymore)

raimund-schluessler, 2019-07-12:
With the update from LibCEC 4.0.1 (LibreElec 8.2.5) to LibCEC 4.0.4 (LibreElec 9.0.x) the PC does not wake up anymore when the TV is powered on. Also the PC does not show up in the CEC-Menu anymore where you could wake up the PC manually.

Used devices:
- TV: Samsung UE55H6470 from 2014
- PC: Intel NUC D34010WYK
- CEC Adapter: P8-INTCECv2  from 2015 running firmware v4

I read in the release notes of 4.0.3 https://github.com/Pulse-Eight/libcec/releases/tag/libcec-4.0.3 that

> The wake up feature is disabled as a result, when a Samsung TV is detected. You can no longer wake up the PC if it's in standby or power it on when the PC is off (with an Intel NUC adapter).

I wonder how I can bring this feature back. Would installing the latest firmware v9 fix the issue?

Turning the PC on is a really useful feature for me, since the NUC is rather hidden and not well accessible in my setup. Disabling the feature for all Samsung TV seems rather drastic, given that older Samsung TVs don't show problems with randomly powering on.

In case you need further information, please tell me.

raimund-schluessler, 2019-07-13:
I have compiled LibreELEC 9.0.2 with libCEC 4.0.1 now and everything works fine. The PC wakes up and is visible in the CEC menu of the TV. So the problem must be within libCEC.

djrm05, 2023-03-18:
> I have compiled LibreELEC 9.0.2 with libCEC 4.0.1 now and everything works fine. The PC wakes up and is visible in the CEC menu of the TV. So the problem must be within libCEC.

@raimund-schluessler could you pls let me know how did you do that? is it working also with Libreelec 11? Any info would be appreciated

closed 2025-02-27 by github-actions[bot]