Pulse-Eight/libcec #464: EventGhost plugin not installed by 4.0.3 on windows?

kbrowder, 2019-07-23:
So I went to install libcec on windows and it looks like there's no eventghost files in the installer, so upgrading from 4.0.2 doesn't actually update eventghost.
I then opened the installers in 7-zip and it's missing the eventghost files (in the 4.0.2 installer it's in the $16 directory)

Any thoughts?

kdschlosser, 2019-12-07:
Sorry for the late reply on this.

The eventghost plugin portion is still a work in progress. You can get the most recent version of the plugin as well as  how to install it from here

http://www.eventghost.net/forum/viewtopic.php?p=49838#p49838

opdenkamp, 2020-04-27:
The plugin will be installed by default in the next version of libcec, if eventghost is found

closed 2020-04-27 by opdenkamp

kdschlosser, 2020-04-27:
@opdenkamp 

I have a newer version of the plugin I can shoot it on over to ya so you can package it up.
We also need to add in some kind of a notice about needing version 0.5 of EventGhost or newer. maybe add a button in the installer so it will install EventGhost for a user that may not have it installed. Up to you. You know the installer script stuff better then I do.

Let me know.

opdenkamp, 2020-04-27:
@kdschlosser if the version is written to the registry or to some sort of file, then it can be auto-detected by the installer

kdschlosser, 2020-04-27:
hmmmm I am going to have to check the installer to see what it adds to the registry. I cannot remember to be honest with ya.

kdschlosser, 2020-04-27:
HKEY_LOCAL_MACHINE\SOFTWARE\Wow6432Node\Microsoft\Windows\CurrentVersion\Uninstall\EventGhost_is1

values MajorVersion and MinorVersion.
if MajorVersion ==  0 and MinorVersion < 5 skip installing the plugin or offer a button to upgrade the EventGhost version.