Pulse-Eight/libcec #467: Unable to send keypress up immediately after keypress down

sparky3387, 2019-08-08:
I am having an issue with sending a short press to an NVIDIA shield, I have a few questions:
1. Is the TV likely the only device able to send remote button app up?
2. Would the Pulse Eight CEC adaptor be able to send txn 08:44:00 txn 08:8b:00 (eg be the initiator 'TV') I already own one of these adaptors, but dont think this would work.
3. Is there an inbuilt way to send mutliple tx's one after another?

Below is the CEC output for the TV remote (when my Raspberry PI CEC device is active source) which can successfully send this command (when the shield is active source):
```
TRAFFIC: [         3516912]     >> 01:44:00
DEBUG:   [         3516912]     >> TV (0) -> Recorder 1 (1): user control pressed (44)
DEBUG:   [         3516912]     SetCurrentButton select (0) D:0ms cur:8b3d68
DEBUG:   [         3516913]     key pressed: select (0) current(ff) duration(0)
DEBUG:   [         3516913]     Changed key select (0) D:0ms cur:ff
DEBUG:   [         3516913]     key pressed: select (0, 0)
DEBUG:   [         3516913]     CheckKeypressTimeout T:67756.367
DEBUG:   [         3516913]     Key select: idle (duration:0) (0) timeout:1939860700ms (rel:500,rep:0,prs:500,rel:0)
TRAFFIC: [         3517095]     >> 01:8b:00
DEBUG:   [         3517095]     >> TV (0) -> Recorder 1 (1): vendor remote button up (8B)
DEBUG:   [         3517096]     key released: select (0) D:183ms
```

If i try to replicate this command I do not get traffic showing request in cec-client and get a feautre abort operation not supported (Recorder 1 is the PI)
```
txn 18:44:00
txn 18:8b:2b
```
OR 
```
tx 18:44:00
tx 18:8b:00
```

The response
```
TRAFFIC: [         2082547]     >> 81:00:8b:00
DEBUG:   [         2082547]     >> Playback 2 (8) -> Recorder 1 (1): feature abort ( 0)
```

If i pretend to be the TV
```
txn 08:44:00
txn 08:8b:00
```

The response
`DEBUG:   [         2236044]     initiator 'TV' is not supported by the CEC adapter
`
If i pretend to be Recorder 2(This device does not exist), I get a response, but still a timeout
```
txn 28:44:00
txn 28:8b:00
```

The response
```
TRAFFIC: [         2012876]     << 28:44:00
WARNING: [         2012966]     unhandled response received: opcode=44 initiator=1 destination=8 response=0
DEBUG:   [         2015876]     command 'user control pressed' timeout
waiting for input
TRAFFIC: [         2016427]     << 28:8b:00
WARNING: [         2016517]     unhandled response received: opcode=8b initiator=1 destination=8 response=0
TRAFFIC: [         2016616]     >> 81:00:8b:00
DEBUG:   [         2016617]     >> Playback 2 (8) -> Recorder 1 (1): feature abort ( 0)
DEBUG:   [         2019427]     command 'vendor remote button up' timeout
```

And lastly broadcast has no response on device or cec-client
```
tx 1f:44:00
tx 1f:8b:00
```

cayfordb, 2019-11-19:
Hi Sparky.   For what it's worth, I have had success sending out CEC Keypress messages within a Kodi script, using cecclient. For a "Pause",  I send out "8B:44:46", wait a moment, and send "8B:45".  Here 8 is the CecTester device port, and B is the Apple TV.    44 is the command to send a keystroke, 46 happens to be the Cec command for "Pause", and 45 is the Upstroke.  It works.

The Kodi framework is this: I turn off cec from the Kodi setup screen.  I have a Kodi script, written in Python, running all the time.   It listens to IR signals from a Tivo Remote Control, and it can send out cec commands via cecclient , and it can talk to other devices on the network such as the Tivo or the AVR. At startup, the script forks off cecclient, using a unix pipe for stdin and a log file for stdout.   The kodi script can then send cec commands by writing to the pipe.   I have a second Kodi script that just reads the logfile output from the cecclient process.  It sees things and tells the main script what it saw, like a TV Power Up, or a TV power down.    The main Kodi script looks for User input from the Tivo remote, and translates that to the keystroke command on other devices like Apple TV or Oppo Bluray, or the Tivo itself.    The Oppo and the Apple accept CEC commands.  Tivo talks telnet.   If I'm watching the Apple TV, and hit Pause on the Tivo Remote, my script sends out the CEC keystroke for Pause, to the Apple TV.   If I'm watching the Bluray, it does the same thing, only sending the CEC command to the blueray.  Ditto on Tivo, but it sends that via telnet.   I mapped about 10 keys for this-- select, up, down, left, right,  pause, play, etc.   The thread that watches the logfile of the CecClient is detecting the CEC addresses so it knows what port to talk to each device, and it recognizes things like the TV turning off and on, so I can power down or up all the other devices.  It's not perfect but it works pretty well.

cayfordb, 2019-11-19:
Also, you MUST have this URL handy:   http://www.cec-o-matic.com

It tells me your "Upstroke" is the "Vendor Remote Button Up".  I would try the Regular Up, which is a 45.

cayfordb, 2019-11-19:
In fact, all the "Feature Abort" errors are complaining about "Vendor Remote UP".  Changing that "8b" to a "45" might be all you need.