Pulse-Eight/libcec #470: cec-utils Ubuntu-Mate 18.04 Raspberry

Obbart, 2019-09-20:
Hi, 
I'm trying to use cec-utils on Raspberry Pi 3 running the arm64 version of Ubuntu Mate.
```
pi@pitest:~$ uname -a
Linux pitest 4.15.0-1047-raspi2 #51-Ubuntu SMP PREEMPT Fri Sep 13 16:39:06 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux
```
but I get the following error:
```
pi@pitest:~$ cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.2
no serial port given. trying autodetect: FAILED
```
and 'cec-client' shows no support for Raspberry
```
pi@pitest:~$ cec-client -l
libCEC version: 4.0.2, compiled on Linux-4.4.0-142-generic ... , features: P8_USB, DRM, P8_detect, randr
Found devices: NONE
```
The only precompiled debs that I can find are for armhf and cannot be installed:
`cec-utils_4.0.4+dfsg1-2+rpi1_armhf.deb   
libcec4_4.0.2+dfsg1-2ubuntu1_arm64.deb
`
I tried to follow the [compile guide](https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md) for Raspberry but in my distro there is no `/opt/vc` directory.

Any suggestions on how to compile on this distro?
Thanks in advance

horich2001, 2020-05-16:
Hi,
I've stumbled on your post while searching for a solution to make cec-utils/libcec to work on RPI4 running Ubuntu  20.04 LTS  arm64. Could not find working binaries with PI option enabled so I had to compile them locally.
First download and build "ARM side libraries used on Raspberry Pi. These typically are installed in /opt/vc/lib and includes source for the ARM side code to interface to: EGL, mmal, GLESv2, vcos, openmaxil, vchiq_arm, bcm_host, WFC, OpenVG."

`git clone https://github.com/raspberrypi/userland.git`
`cd userland/`
`./buildme --aarch64`

Then build the libCEC following the [compile guide](https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md) mentioned by you above
I've changed the cmake command to include enable   Raspberry Pi support:

  `cmake   -DHAVE_RPI_API=1  -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..`

had to  do this trick to make /opt/vc/include visible to the include path of the compiler:
`sudo ln -s /opt/vc/include/ /usr/local/include/aarch64-linux-gnu`

also I had to add /opt/vc/lib to the LIBRARY_PATH:
` /etc/ld.so.conf.d/aarch64-linux-gnu.conf`

final issue was that I could not run cec-client as normal user (failed to open vchiq instance)
Had to fix the permission of:
`crw------- 1 root root 237, 0 Apr  1 17:23 /dev/vchiq`

`root@ubuntu:~# chown root:video /dev/vchiq` 
`root@ubuntu:~# chmod  g+rw /dev/vchiq`

Now all looks good:
`ubuntu@ubuntu:~$ cec-client -l
libCEC version: 5.0.0, git revision: libcec-4.0.4+61-4ef0486, compiled on 2020-05-16 18:38:38 by ubuntu@ubuntu on Linux 5.4.0-1008-raspi (aarch64), features: P8_USB, DRM, P8_detect, randr, RPi, AOCEC
Found devices: 1

device:              1
com port:            RPI
vendor id:           2708
product id:          1001
firmware version:    1
type:                Raspberry Pi`

`ubuntu@ubuntu:~$ cec-client -b
No device type given. Using 'recording device'
CEC Parser created - libCEC version 5.0.0
no serial port given. trying autodetect: 
 path:     Raspberry Pi
 com port: RPI

opening a connection to the CEC adapter...
DEBUG:   [             346]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [             347]	Open - vc_cec initialised
DEBUG:   [             347]	logical address changed to Free use (e)
NOTICE:  [             347]	connection opened
DEBUG:   [             347]	processor thread started
DEBUG:   [             347]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [             347]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [             347]	<< e0
DEBUG:   [             377]	>> POLL sent
DEBUG:   [             377]	TV (0): device status changed into 'present'
...........
`
I hope it helps. Have fun! 

Cheers,
Horia

clementnuss, 2021-04-28:
Hi @horich2001 !

It helps a lot, was just able to compile this on RPi4 Ubuntu 20.04.2 👏

igox, 2022-01-18:
Hi @horich2001 ,

It helped a lot. 
I did succeed to compile and have the lib/binaries on Debian 10 Arm64.

But when I do `pi@hyperhdr:~$ cec-client -l`, `cec-client` command fails with the following message:
```libCEC version: 4.0.7, git revision: libcec-4.0.7, compiled on 2022-01-18 12:43:02 by pi@hyperhdr on Linux 5.10.63-v8+ (aarch64), features: P8_USB, DRM, P8_detect, randr, RPi
assertion failure:/build/raspberrypi-userland-Y644aj/raspberrypi-userland-2+git20211125~155417+14b90ff/interface/vmcs_host/vc_vchi_dispmanx.c:201:vc_vchi_dispmanx_init():success == 0
```

I do suspect something going wrong with the "_ARM side libraries used on Raspberry Pi_"...
But not sure about it and where to start digging...

Best regards,
iGoX

closed 2025-02-27 by github-actions[bot]