Pulse-Eight/libcec #476: RPI 4: Buster, cec-client 4.0.4 segmentation fault

schadha101, 2019-10-05:
uname -a
Linux 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux

cec-client --info
libCEC version: 4.0.4, compiled on Linux-4.15.0-48-generic ... , features: P8_USB, DRM, P8_detect, randr, RPi, Exynos, AOCEC

Followed these steps to download and compile cec:
apt-get update
apt-get install cmake libudev-dev libxrandr-dev python-dev swig
git clone https://github.com/Pulse-Eight/libcec.git
mkdir libcec/build
cd libcec/build
cmake ..
make -j4
sudo make install
sudo ldconfig

 echo "on 0" | cec-client -s -d 2
opening a connection to the CEC adapter...
WARNING: [            3322]     unhandled response received: opcode=0 initiator=e destination=0 response=0
WARNING: [           17104]     unhandled response received: opcode=84 initiator=1 destination=f response=0
WARNING: [           18635]     unhandled response received: opcode=47 initiator=1 destination=0 response=0
WARNING: [           20165]     unhandled response received: opcode=47 initiator=1 destination=0 response=0
WARNING: [           21695]     unhandled response received: opcode=8f initiator=1 destination=0 response=0
WARNING: [           23228]     unhandled response received: opcode=8f initiator=1 destination=0 response=0
WARNING: [           24759]     unhandled response received: opcode=8c initiator=1 destination=0 response=0
WARNING: [           26289]     unhandled response received: opcode=8c initiator=1 destination=0 response=0
WARNING: [           27819]     unhandled response received: opcode=4 initiator=1 destination=0 response=0
Segmentation fault

NOTE:
I tried other commands such as standby, as, pow all produce segmentation fault after completing
echo 'standby 0' | cec-client -s -d 2
echo 'as' | cec-client -s -d 2                          
echo 'scan' | cec-client -s -d 2

brianmduncan, 2020-01-03:
I can confirm the same behavior.  Is no one maintaining this project anymore?

uname -a
Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

cec-client --info
libCEC version: 4.0.4, compiled on Linux-4.15.0-48-generic ... , features: P8_USB, DRM, P8_detect, randr, RPi, Exynos, AOCEC

No matter what I do it seems to complete most commands but exits in a segmentation fault:

echo "on 0" | cec-client -s -d 1
opening a connection to the CEC adapter...
Segmentation fault

ivanek666, 2020-01-05:
Same issue..

zocker-160, 2020-03-02:
do you have your monitor in HDMI0 or HDMI1 ?

I had the same issue when plugging the monitor / TV into HDMI1, after plugging that into HDMI0 I had no issues since.

ivanek666, 2020-03-05:
Same, after pluggin into HDMI0 and stopped using hdmi to hdmi mini adapter all worked..

drftg, 2020-05-23:
Same problem on all 3 HDMI inputs of Samsung 32K5600AW. On RPi 3B+ and RPi 4
Linux pi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
libCEC version: 4.0.4, compiled on Linux-4.15.0-48-generic ... , features: P8_USB, DRM, P8_detect, randr, RPi, Exynos, AOCEC

pi@pi:~$ echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Unknown
osd string:    TV
CEC version:   unknown
power status:  unknown
language:      ???

device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
Segmentation fault

opdenkamp, 2020-07-10:
can you check whether this still happens when using libCEC 6.0.0

drftg, 2020-08-04:
@opdenkamp Sorry for the late response. Only now do I have the machine back for maintenance. I tried libCEC 6.0.2 on RPi 3B+

 cat /proc/device-tree/model
Raspberry Pi 3 Model B Rev 1.2

cec-client --info
libCEC version: 6.0.2, git revision: libcec-6.0.2+1-76551ea, compiled on 2020-08-04 21:35:14 by pi@iScored on Linux 4.19.97-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr

but now I get autodetect failed

 echo 'scan' | cec-client -s -d 1
autodetect FAILED

I have no idea if the Segmentation fault is gone

opdenkamp, 2020-08-05:
your build doesn't include support for the Pi
this page describes how to compile it with support for the Pi: https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md

drftg, 2020-08-05:
Sorry about that! Forgot to include platform.

cec-client --info
libCEC version: 6.0.2, git revision: libcec-6.0.2+1-76551ea, compiled on 2020-08-05 09:00:35 .... on Linux 5.4.51-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi

echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
ERROR:   [             127]     Open - vc_cec could not be initialised
ERROR:   [             127]     could not open a connection (try 1)
.
.
.
ERROR:   [            9137]     could not open a connection (try 10)
unable to open the device on port RPI
ERROR:   [           10137]     could not start CEC communications

---------------
But the segmentation fault is gone!

nperez0111, 2020-08-12:
Still is seg faulting for me:

`uname -a`
Linux nickthesickpi 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux

`cec-client --info`

libCEC version: 6.0.2, git revision: libcec-6.0.2+1-76551ea, compiled on 2020-08-12 01:05:27 by pi@nickthesickpi on Linux 5.4.51-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi

Getting this output:
```
echo 'scan' | cec-client -s -d 1                                                                                                                              pi@nickthesickpi
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
[1]    8807 done                echo 'scan' | 
       8808 segmentation fault  cec-client -s -d 1```

nperez0111, 2020-08-12:
Actually, after the recompile and fresh boot with the hdmi plug on HDMI0 on the PI, it worked for me.

Why would it need a reboot I don't know?

lderossi, 2021-02-18:
I'm having a similar RPi4 issue on the latest from git (6.0.2).

- Raspberry Pi 4 Model B Rev 1.2
- Linux raspberrypi 5.10.14-v7l+ #1401 SMP Mon Feb 8 14:27:07 GMT 2021 armv7l GNU/Linux
- libCEC version: 6.0.2, git revision: libcec-6.0.2+1-76551ea, compiled on 2021-02-18 17:26:33 by pi@raspberrypi on Linux 5.10.14-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
- Connected to a Samsung TV on HDMI0 (closest to USB power)

I see segmentation faults, typically after the command has finished; I can get a scan, and can turn on/off the TV I'm connected to. But execution is very slow and has lots of command timeouts in the debug log. Running on a Pi3B connected to the same TV via the same cable (though the Pi4 has an adapter, I've also tried two different direct micro HDMI->HDMI from Pi4 to TV as well) works flawlessly and almost instantaneously.

On my Pi3B turning off the tv with "echo 'standby 0.0.0.0' | cec-client -s" takes about 2 seconds. On the Pi4 it can take 15-30.

With no connection on either HDMI port, running "echo 'scan' | cec-client -s" takes over a minute. Here's the output from a scan with no HDMI devices connected:

` echo 'scan' | cec-client -s -d 31
opening a connection to the CEC adapter...
DEBUG:   [              82]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              82]     Open - vc_cec initialised
DEBUG:   [              82]     logical address changed to Free use (e)
NOTICE:  [              82]     connection opened
DEBUG:   [              82]     processor thread started
DEBUG:   [              82]     << Broadcast (F) -> TV (0): POLL
DEBUG:   [              83]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [              83]     << e0
DEBUG:   [            1083]     command 'feature abort' timeout
DEBUG:   [            1583]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [            1583]     << e0
DEBUG:   [            1614]     >> POLL not sent
DEBUG:   [            1614]     TV (0): device status changed into 'not present'
DEBUG:   [            1614]     registering new CEC client - v6.0.2
DEBUG:   [            1614]     SetClientVersion - using client version '6.0.2'
NOTICE:  [            1614]     setting HDMI port to 1 on device TV (0)
DEBUG:   [            1614]     << Broadcast (F) -> TV (0): POLL
DEBUG:   [            1614]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [            1614]     << e0
DEBUG:   [            2615]     command 'feature abort' timeout
DEBUG:   [            3115]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [            3115]     << e0
DEBUG:   [            3146]     >> POLL not sent
DEBUG:   [            3147]     SetConfiguration: double tap timeout = 200ms, repeat rate = 0ms, release delay = 500ms
DEBUG:   [            3147]     detecting logical address for type 'recording device'
DEBUG:   [            3147]     trying logical address 'Recorder 1'
DEBUG:   [            3147]     << Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [            3148]     << 11
TRAFFIC: [            8735]     << 11
DEBUG:   [           13331]     >> POLL not sent
DEBUG:   [           13331]     using logical address 'Recorder 1'
DEBUG:   [           13331]     Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [           13331]     Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [           13331]     Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [           13332]     Recorder 1 (1): CEC version 1.4
DEBUG:   [           13332]     AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [           13332]     logical address changed to Recorder 1 (1)
DEBUG:   [           13332]     Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [           13332]     Recorder 1 (1): menu language set to 'eng'
DEBUG:   [           13333]     GetPhysicalAddress - physical address = 1000
DEBUG:   [           13333]     using auto-detected physical address 1000
DEBUG:   [           13333]     Recorder 1 (1): physical address changed from ffff to 1000
DEBUG:   [           13333]     << Recorder 1 (1) -> broadcast (F): physical address 1000
TRAFFIC: [           13333]     << 1f:84:10:00:01
DEBUG:   [           14333]     command 'report physical address' timeout
TRAFFIC: [           14834]     << 1f:84:10:00:01
NOTICE:  [           14864]     CEC client registered: libCEC version = 6.0.2, client version = 6.0.2, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0, git revision: libcec-6.0.2+1-76551ea, compiled on 2021-02-18 17:26:33 by pi@raspberrypi on Linux 5.10.14-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
DEBUG:   [           14864]     << Recorder 1 (1) -> TV (0): OSD name 'CECTester'
DEBUG:   [           14864]     << Recorder 1 (1) -> TV (0): POLL
TRAFFIC: [           14864]     << 10
DEBUG:   [           15865]     command 'feature abort' timeout
TRAFFIC: [           16365]     << 10
WARNING: [           16396]     unhandled response received: opcode=84 initiator=1 destination=f response=0
DEBUG:   [           17365]     command 'feature abort' timeout
DEBUG:   [           17865]     >> POLL not sent
[... 145 lines cut ...]
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
[... cut ...]
DEBUG:   [           65990]     Recorder 1 (1): CEC version unknown
DEBUG:   [           65990]     Recorder 1 (1): osd name set to 'Recorder 1'
DEBUG:   [           65990]     Recorder 1 (1): device status changed into 'unknown'
DEBUG:   [           65990]     unregistering all CEC clients
DEBUG:   [           66092]     UnregisterLogicalAddress - releasing previous logical address
Segmentation fault
`

Any thoughts on what I can do to debug further? I've fiddled with most of the hdmi settings in config.txt, rebuilt and rebooted many times.

lderossi, 2021-02-18:
I happen to have a Pulse Eight USB adapter, and connected that to the Pi4 and inserted it into the HDMI cable path between the Pi4 and the Samsung TV. Running cec-client pointing to the USB adapter works as expected; no segfaults, and commands are as quick as I'd expect.

ethaniel, 2023-02-27:
I have a segfault on a freshly built  (via instructions above) libcec `6.0.2` on rpi4 and a Samsung TV `QA43Q65BAKXXT`:
```
libCEC version: 6.0.2, git revision: libcec-6.0.2+3-bf5a97d, compiled on 2023-02-27 18:06:15 by pi@ampur on Linux 5.10.17-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
```

Here is the log:
```
pi@ampur:~ $ echo 'scan' | cec-client -s -d 7
opening a connection to the CEC adapter...
NOTICE:  [              68]	connection opened
NOTICE:  [            1599]	setting HDMI port to 1 on device TV (0)
NOTICE:  [           14842]	CEC client registered: libCEC version = 6.0.2, client version = 6.0.2, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0, git revision: libcec-6.0.2+3-bf5a97d, compiled on 2023-02-27 18:06:15 by pi@ampur on Linux 5.10.17-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
WARNING: [           16372]	unhandled response received: opcode=84 initiator=1 destination=f response=0
requesting CEC bus information ...
WARNING: [           20963]	unhandled response received: opcode=0 initiator=1 destination=0 response=1
WARNING: [           22493]	unhandled response received: opcode=0 initiator=1 destination=0 response=1
WARNING: [           24024]	unhandled response received: opcode=0 initiator=1 destination=0 response=1
WARNING: [           25554]	unhandled response received: opcode=0 initiator=1 destination=0 response=1
WARNING: [           27085]	unhandled response received: opcode=0 initiator=1 destination=2 response=1
WARNING: [           28615]	unhandled response received: opcode=0 initiator=1 destination=2 response=1
WARNING: [           30145]	unhandled response received: opcode=0 initiator=1 destination=3 response=1
WARNING: [           31676]	unhandled response received: opcode=0 initiator=1 destination=3 response=1
WARNING: [           33206]	unhandled response received: opcode=0 initiator=1 destination=4 response=1
WARNING: [           34736]	unhandled response received: opcode=0 initiator=1 destination=4 response=1
WARNING: [           36266]	unhandled response received: opcode=0 initiator=1 destination=5 response=1
WARNING: [           37797]	unhandled response received: opcode=0 initiator=1 destination=5 response=1
WARNING: [           39327]	unhandled response received: opcode=0 initiator=1 destination=6 response=1
WARNING: [           40858]	unhandled response received: opcode=0 initiator=1 destination=6 response=1
WARNING: [           42388]	unhandled response received: opcode=0 initiator=1 destination=7 response=1
WARNING: [           43918]	unhandled response received: opcode=0 initiator=1 destination=7 response=1
WARNING: [           45448]	unhandled response received: opcode=0 initiator=1 destination=8 response=1
WARNING: [           46979]	unhandled response received: opcode=0 initiator=1 destination=8 response=1
WARNING: [           48509]	unhandled response received: opcode=0 initiator=1 destination=9 response=1
WARNING: [           50039]	unhandled response received: opcode=0 initiator=1 destination=9 response=1
WARNING: [           51570]	unhandled response received: opcode=0 initiator=1 destination=a response=1
WARNING: [           53100]	unhandled response received: opcode=0 initiator=1 destination=a response=1
WARNING: [           54630]	unhandled response received: opcode=0 initiator=1 destination=b response=1
WARNING: [           56161]	unhandled response received: opcode=0 initiator=1 destination=b response=1
WARNING: [           57691]	unhandled response received: opcode=0 initiator=1 destination=c response=1
WARNING: [           59222]	unhandled response received: opcode=0 initiator=1 destination=c response=1
WARNING: [           60752]	unhandled response received: opcode=0 initiator=1 destination=d response=1
WARNING: [           62282]	unhandled response received: opcode=0 initiator=1 destination=d response=1
WARNING: [           63813]	unhandled response received: opcode=0 initiator=1 destination=e response=1
WARNING: [           65343]	unhandled response received: opcode=0 initiator=1 destination=e response=1
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
NOTICE:  [           65955]	unregistering client: libCEC version = 6.0.2, client version = 6.0.2, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0, git revision: libcec-6.0.2+3-bf5a97d, compiled on 2023-02-27 18:06:15 by pi@ampur on Linux 5.10.17-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
Segmentation fault
```

ethaniel, 2023-02-28:
Problem was solved for me when I upgraded from `buster` to `bullseye` (I'm still using the manually compiled `libCEC version: 6.0.2` from the instructions above). All cec commands perform immediately:

```
pi@ampur:~ $ echo 'on 0' | cec-client -s -d 7
opening a connection to the CEC adapter...
NOTICE:  [              86]	connection opened
NOTICE:  [              87]	setting HDMI port to 1 on device TV (0)
NOTICE:  [              89]	CEC client registered: libCEC version = 6.0.2, client version = 6.0.2, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0, git revision: libcec-6.0.2+3-bf5a97d, compiled on 2023-02-27 18:06:15 by pi@ampur on Linux 5.10.17-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
NOTICE:  [              90]	<< powering on 'TV' (0)
NOTICE:  [              90]	unregistering client: libCEC version = 6.0.2, client version = 6.0.2, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0, git revision: libcec-6.0.2+3-bf5a97d, compiled on 2023-02-27 18:06:15 by pi@ampur on Linux 5.10.17-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
```

closed 2025-02-27 by github-actions[bot]