Pulse-Eight/libcec #482: LG Magic remote doesnt work

charta77, 2019-12-03:
Hi,
I have LG TV (series 9000) connected to RPi2 with Kodi(Libreelec). HDMI cable is Audioquest Pearl.
I want use LG Magic remote to control Kodi, but It doesnt work, any button. But in generated [log](https://paste.kodi.tv/noheronizu.kodi) file I see some communication via CEC.
f.e.:
`2019-11-27 13:27:22.180 T:1600844672 WARNING: CecLogMessage - FIXME: LG seems to have bugged out. resetting to 'in transition standby to on'. the return button will not work`
Could someone help me please?

thank a lot
Petr

kdschlosser, 2019-12-07:
CEC is a half way done protocol. There really is no requirements or certification of a device that uses CEC. So what happens is only a small part of the protocol gets used. or they use a portion of the protocol and then "add" their own pieces to it. the remote control portion seems to be one of the most common things that is left out or changed.

charta77, 2019-12-07:
thank you for your reply. No, I didnt try cecClient. is there any cookbook how to use it with Kodi on Libreelec? thank you!
Petr

kdschlosser, 2019-12-07:
well we need to know if the remote actually works over CEC or not. bcause if it does not work then doing anything with Kodi is going to be a waste of time.
Is your CEC adapter built onto a device (like a Raspberry Pi) or is it a USB version?

charta77, 2019-12-07:
as I wrote I want to control Kodi(RPi2/Libreelec) with my TV's remote cotrol (LG 9000 series Magic control). Now I use to control Kodi remote MeLE F10.
But my dream is I have only one remote for TV and Kodi -> LG Magic control

kdschlosser, 2019-12-07:
OK so I still need to have this question answered. Not all Raspberry Pi's come with CEC built into it. 
So is the CEC built into the Raspberry Pi or do you have a separate dongle plugged into the USB that handles the CEC?

charta77, 2019-12-07:
CEC is built into the RPi2.
thank you for your time

kdschlosser, 2019-12-08:
no worries m8. ok so no option to run windows.

I can hammer out a python script you can run which will provide really detailed logs. give me a day or so to get that together for ya so we can see exactly what is happening.

charta77, 2019-12-08:
you are great! thank you very much

charta77, 2020-01-21:
I would like to ask you about the script - is there any good news?
thanks
Petr

charta77, 2020-01-26:
@kdschlosser Hi,
is there any chance to do the script?
thank you

charta77, 2020-02-17:
@kdschlosser Hi Kevin, is something new about the script? thanks

opdenkamp, 2020-03-28:
Sorry for the delay. LG's magic remote does not use CEC. It can only control the TV.
As for that warning: that one can show up if the HDMI port isn't set up correctly in Kodi. You can configure it in Kodi, or move to HDMI 1 of the TV (default setting)

closed 2020-03-28 by opdenkamp

charta77, 2020-03-30:
@opdenkamp thank you for your reply.
If the magic remote doesnot uses CEC, why my TV (9000 series) support Simplink(hdmi-cec)? So I think some cec communication between TV and Kodi may exists... thank you and have a nice day
Petr

opdenkamp, 2020-03-30:
Because your TV also supports the standard IR remote, and you can use features like selecting an HDMI input.

charta77, 2020-03-30:
ok,thank you

charta77, 2020-04-13:
@opdenkamp  It works :)

jekkos, 2020-11-02:
@charta77 How did you get it to work in the end?

charta77, 2020-11-03:
@jekkos Hi,
simply via device connector. The Kodi is defaultly set as blu ray so set it as "other box".

jekkos, 2020-11-03:
It seems my CEC link is working in some way (it IS turning on my LG beamer) , also tried to echo ON and pipe it to cec-client, which also works... However no buttons whatsoever.

Do you have it directly connected to your HDMI on raspberry pi or do you use an extra device in between? Do you use a special remote.xml? If yes would you mind to share?

charta77, 2020-11-03:
RasPi is connected directly (hdmi 2.0 cable), without special remote.xml

FireEmerald, 2020-11-12:
> Sorry for the delay. LG's magic remote does not use CEC. It can only control the TV.
> As for that warning: that one can show up if the HDMI port isn't set up correctly in Kodi. You can configure it in Kodi, or move to HDMI 1 of the TV (default setting)

It is possible to control Kodi with your Magic Remote on a LG smart TV.
My issue was a wrong configuration inside the "Home Dashboard" of the LG smart TV.

**How to reset the configured device in your LG smart TV:**
> Ok, first, when I connected the HDMI cable to the TV, the raspberry was identified like a Roku and categorized in the Set-Top Box/OTT group. The remote control was not able to interact with raspberry. Simplink (HDMI-CEC) was ON in the connection settings options.

> Some time later, after several attempts, and after a new try removing and reconnecting the raspberry HDMI cable in another port, I realized in just one of the categories listed in the Device Connector page, where you can find: Set-Top Box/OTT, Game Console, Home Theater etc, exactly in the Set-Top Box/OTT, an interesting button, the reset device button. So, I just clicked on it and got out that page and returned to Raspberry input to check if I had success, and for my happiness, yes, remote control working.

I found the solution here: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=231676#p1637861

mlindner, 2021-01-13:
> Sorry for the delay. LG's magic remote does not use CEC. It can only control the TV.

This is not true. It can control other devices perfectly fine. If you mean the "magic mouse" part of the controller no, that can't be used, but button presses go across CEC from the TV without issue.

malard, 2021-01-13:
Sounds like they improved it since I last looked then!

jekkos, 2021-01-13:
I'm quite convinced that it does have some CEC capabilities, however I was unable to make it work and gave up. Eventually I went with good old gpio receiver on the rpi that now uses my amplifier's remote (2 in one), which is even nicer in fact.