Pulse-Eight/libcec #489: RPI 4 libCEC not detecting TV on HDMI 1

mateorotim, 2020-01-17:
I'm using raspbian buster full image, I have compiled libCEC for raspberry, when i connect 2 different TV's to raspberry with the same cables libCEC only detects the TV that is connected to HDMI 0. So it is not the cables or the TV's because when i switch the HDMI inputs on raspberry libCEC detects other TV on HDMI 0 and nothing on HDMI 1.

Any idea how to resolve this? I need to be able to control both TV's.

zocker-160, 2020-03-02:
the CEC protocol only supports one TV / monitor at a time, which is a hard limit in the spec

opdenkamp, 2020-04-01:
only one hdmi output on the pi4 is supported. if you want to control the second screen too, then you'll need a usb-cec adapter: https://www.pulse-eight.com/p/104/usb-hdmi-cec-adapter and indeed start a second instance of libCEC

closed 2020-04-01 by opdenkamp

popcornmix, 2020-04-03:
The KMS driver which is being developed does expose cec devices for each hdmi display.
In combination with https://github.com/Pulse-Eight/libcec/pull/380 it should allow cec control of each display. CEC support in KMS is not working yet, so you'll have to wait a little to get a definite answer.

opdenkamp, 2020-04-04:
I was indeed talking about the current implementation. Thanks for the update!

audas, 2020-11-10:
Hi this has been a fair while and I have tested this with the new kernel and we are not able to work with the second port. Engineers at Raspi have suggested this is something that libcec will have to address - is there any chance of this being implemented, and if so any time soon ?