Pulse-Eight/libcec #490: using cec-client blocks blu ray player / sound system

Maxus1998, 2020-01-25:
I have tried the cec-client several times now, but I always get the same problem.
As soon as I listen for commands or send them with the client, my Home Theater System (LG 2.1 Audo + Blu Ray Player Combo) won't react to the commands of my TV any more.

So normally when I press volume up/down or turn my TV on or off the Blu Ray Player changes volume or turns on or off.
When I start my TV first and adjust the volume, it displays it is sending those commands to the connected device. If I then activate the cec-client, it just changes the volume of the TV.

My setup:
Samsung UE32F6470
LG BH7240C 2.1 Smart Blu Ray Player
Raspberry Pi 3

Raspberry<---HDMI--->Blu Ray Player<---HDMI--->TV

Is there some special action I did forget to perform, is my setup just wrong or am I completly missing something?

opdenkamp, 2020-04-01:
I'll have to see the traffic log to see what's going on. Please upload the full output of cec-client.

xammmue, 2025-02-26:
I don't have the setup for testing anymore which means I am no longer affected by the supposed problem
Nobody else reported anthing so I'll just close that then

closed 2025-02-26 by xammmue