Pulse-Eight/libcec #497: CEC-adapter sending signal when Wake-on-USB is enabled

snixel, 2020-03-30:
I use WOL to turn on my HTPC. Unfortunately when I enable WOL in BIOS it also enables Wake-on-USB (there is no way to have one enabled without the other). Because of this my CEC adapter gets power on it's USB port causing it to send out a HDMI active signal even if my HTPC is powered off.
I was wondering if this could be solved through firmware?

opdenkamp, 2020-04-07:
The CEC adapter doesn't send any signal to power on the TV by itself. Do you have a Samsung TV perhaps? If so, please take a look at this ticket and the v11 that I've just uploaded there: https://github.com/Pulse-Eight/libcec/issues/459#issuecomment-609795940

snixel, 2020-04-13:
The problem was not that my TV would turn on automatically, I have it connected to my AVR. The problem was that when my HTPC with the CEC adapter was powered off and I would turn on another CEC enabled device also connected to my AVR, my AVR would start flipping between the input of the powered on device and the input where the CEC adapter was connected (even when the HTPC it was connected to was completely powered off).
I troubleshooted the problmem and came to the conclusion that the CEC adapter still received power because I enabled WOL (and thus also Wake-on-USB) in the BIOS of the HTPC. Because of this the CEC adapter still send out some signal causing my AVR to think there was a powered on device on the input. When I unplugged the USB cable from the CEC adapter or when I disabled wake-on-USB the problem dissapeared.
This might be some bad CEC iplementation of my AVR , I don't know for sure (it's a Marantz receiver).

But, the good news is that with firmware v11 it no longer does this :) I have thoroughly tested it over the weekend and my AVR no longer flips inputs even when WOL is enabled on my HTPC.
So a big **THANK YOU** for your hard work Lars!!!! I have had this issue for almost 3 years now and you just fixed it! 🥰

closed 2020-05-12 by snixel

opdenkamp, 2020-05-12:
Sorry, I missed your comment. Thanks for reporting back and good to hear that it's been fixed.