Pulse-Eight/libcec #500: pyCecClient.py Error module 'cec' has no attribute 'libcec_configuration' --  Ubuntu 19.10

Electrik-rich546456, 2020-04-06:
Hi I am really very new at all this and I have managed to teach myself as much as i can but I have a few issues I cannot find how to fix...

First question is if i compile this as per [these instructions](https://github.com/Pulse-Eight/libcec/blob/master/docs/README.linux.md)

and then if I compile using these [instructions](https://github.com/Pulse-Eight/libcec/blob/master/docs/README.debian.md) what happens to the first compiled cec.so (or cec.py or whatever extension) file used in python?

If i run `sudo pip freeze` I don't see cec but if i run `pip freeze` i see `cec==0.2.7`
If I run `pip show cec`I get `.local/lib/python2.7/site-packages` as its location.
even though one of the compile output lines was `Installing: /usr/local/lib/python3.7/dist-packages/cec.py`
What is going on??
why is system not using new one??

That leads me on to my next question

Where does the [instructions](https://github.com/Pulse-Eight/libcec/blob/master/docs/README.debian.md) fit into the main set.

'apt-get update
apt-get install cmake libudev-dev libxrandr-dev python-dev swig
git clone https://github.com/Pulse-Eight/libcec.git
mkdir libcec/build
cd libcec/build'

Does it go here ???????

'cmake ..
make -j4
sudo make install
sudo ldconfig'

jagofulford, 2020-11-10:
I'm also having the same issue. Did you find a resolution for this?

mtullo27, 2023-02-27:
@jagofulford @Electrik-rich546456 I am also having this issue, have you found a solution?

jagofulford, 2023-02-28:
I unfortunately never did find a fix.

drolevar, 2023-04-07:
Guys, you are all using cec which is installed via pip, it's a different library: https://github.com/trainman419/python-cec
This script uses bindings built by libcec itself.

So you have to **uninstall** it via `pip uninstall cec` and then try running this script. If it doesn't, try building libcec from the source and installing it with sudo make install.

PR #655 (2025-02-14): fix python client
fixes #500, #649, #629
closed 2025-02-14 by opdenkamp