Pulse-Eight/libcec #502: Kodi does not shutdown

Enominus, 2020-04-07:
Hi all,

thx for this great feature. I have a pulse eight hdmi cec adapter, a win 10 HTPC with the latest Kodi and a LG Beamer. In general, everything works fine - I am able to control kodi with my LG remote. But I have one problem: When I configure in kodi settings "When TV ist switched of: shutdown", Kodi does not shutdown. Kodi instead quits itselfs, the win 10 desktop appears. What works: When I configure "standby", Kodi goes correctly in standby mode. But I dont like the standby function. More over I would like to use the shutdown function. Is this a known bug? Is there a known workaround? 

thx in advance!
Enominus

Sudrien, 2020-04-09:
https://wiki.gentoo.org/wiki/Kodi currently has a udev/polkit rule at the bottom of the page to allow this

Enominus, 2020-04-09:
Hi Sudrien,

thanks for your reply! Unfortunately I'm a rookie with very less experience. Could you please give me a step by step instruction how I could solve my problem?

thx!

Sudrien, 2020-04-10:
Not having a windows machine available, I can direct you to https://forum.kodi.tv/showthread.php?tid=217477 - it seems to match your description.

Enominus, 2020-04-10:
yea that helped! Thanks for your support!

closed 2020-04-10 by Enominus