Pulse-Eight/libcec #507: Cannot update firmware to v12 on Win10

strim, 2020-05-07:
On attempt to execute cec-firmware-v12.exe with admin permissions, the application error reported:
```
 Installing required drivers
 * Pulse-Eight USB to CEC Adapter d
 * Pulse-Eight USB to CEC Adapter d

 Upgrading the firmware.

 DO NOT UNPLUG THE ADAPTER WHILE TH

 flash.exe - Application Error
 The appliaction was unable to start correctly (0xc000007b).
 Click OK to close the application
```
Sounds like flash.exe binary can not start.
[image]

strim, 2020-05-09:
I checked the other issues and found the same problem reported here in the thread:
https://github.com/Pulse-Eight/libcec/issues/459#issuecomment-607845372
Will try to install Microsoft Visual C++ 2010 Redistributable Package (x86)

strim, 2020-05-09:
Confirmed.
Was able to install v12 after installing [MS Visual C++ 2010 Redistributable Package (x86)|(https://www.microsoft.com/en-us/download/details.aspx?id=5555)]

closed 2020-05-09 by strim

satmandu, 2020-05-18:
@opdenkamp Please add a check for MS Visual C++ 2010 Redistributable Package (x86) installation when running this program. (Some of us only open a virtual machine install of Windows 10 for firmware updates, and may not have this installed by default.)

pitts-mo, 2020-05-28:
Yes, I too have found a check for this dependency would be helpful. 

My reason, I use linux and attempted:
cec-firmware-v12.exe update in a win10 64bit VM and hit this error.
cec-firmware-v12.exe update in fresh install of win10 on extra hard drive and hit this error.
cec-firmware-v9.exe update in fresh install of win10 on extra hard drive and had success.

Thanks to all libcec dev contributors for the great work. :-)

satmandu, 2020-05-28:
I was able to get cec-fimware-v12.exe to work from a mac running win10 on a Virtualbox VM. Just make sure that the USB device (both of them, since the ID changes during the update, so look for that during the update) are connected to the VM.