Pulse-Eight/libcec #511: Panasonic Viera TH-50PX600A - Power On does not work

heitbaum, 2020-05-20:
Could you please assist in sharing the CEC code that will allow the TV to power on / identify why lib CEC is not successful in powering this device on using hex 04.

The following trace (From the NUC $8) shows the AppleTV $4 which became the active source when I used the Apple remote. Trying to turn the TV on. And it always errors. The second trace shows the cec-client trying to turn the TV on. Same issue - with “feature abort”

Standby 0 works fine.

I tried other codes whilst in standby mode (http://www.cec-o-matic.com/) with no success.

The manual indicates it works. But without another Panasonic device - I can’t validate...
https://www.manualslib.com/manual/118537/Panasonic-Viera-Th-42px600a.html?page=17#manual

TRAFFIC: [           64692]     >> 40:04
DEBUG:   [           64692]     TV (0): power status changed from 'standby' to 'in transition from standby to on'
TRAFFIC: [           64796]     >> 4f:82:11:00
DEBUG:   [           64796]     Playback 1 (4): physical address changed from ffff to 1100
DEBUG:   [           64796]     Playback 1 (4): power status changed from 'unknown' to 'on'
DEBUG:   [           64796]     making Playback 1 (4) the active source
DEBUG:   [           64796]     >> Playback 1 (4) -> Broadcast (F): active source (82)
TRAFFIC: [           64931]     >> 04:00:04:01
TRAFFIC: [           65034]     >> 40:8e:00

==

on 0
NOTICE:  [          699001]     << powering on 'TV' (0)
TRAFFIC: [          699001]     << 80:04
DEBUG:   [          699062]     TV (0): power status changed from 'in transition from on to standby' to 'in transition from standby to on'
TRAFFIC: [          699533]     >> 08:00:04:01
DEBUG:   [          699533]     >> TV (0) -> Playback 2 (8): feature abort ( 0)
on 0
NOTICE:  [          710747]     << powering on 'TV' (0)
TRAFFIC: [          710747]     << 80:04
TRAFFIC: [          711277]     >> 08:00:04:01
DEBUG:   [          711277]     >> TV (0) -> Playback 2 (8): feature abort ( 0)

heitbaum, 2020-05-23:
So on much searching and testing. The code required to power on the Panasonic is the same as the AMP.
In my case: (Playback 2 —> TV)
80:44:40
80:44

So the question now is, how can libcec be updated to run this CEC to their TV?

CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Panasonic
osd string:    TV
CEC version:   unknown
power status:  on
language:      eng

ver 0
CEC version unknown
DEBUG:   [         3317615]     'get cec version' is marked as unsupported feature for device 'TV'
ven 0
vendor id: 008045

heitbaum, 2020-10-31:
https://forum.libreelec.tv/thread/22894-cec-ctl-on-generic/#wcf29

heitbaum, 2020-10-31:
I have been testing with the kernel cec driver.
Compile the kernel with the CONFIG_USB_PULSE8_CEC=m

# Then run
modprobe -v pulse8_cec
/storage/inputattach --pulse8-cec /dev/ttyACM0 --daemon
cec-ctl -playback

**OFF**
> `cec-ctl --standby -t0

**ON**
> cec-ctl --custom-command=cmd=0x44,payload=0x40 -t0
> cec-ctl --custom-command=cmd=0x45 -t0

audas, 2020-11-09:
trying to work out how to use cec with kernel on raspberry pi. The branch 5.9 can control the second HDMI port via cec however I have not been able to get it working.

The was a vague comment left on a forum which said you can only use the kernel to control the second hdmi port via cec - however I have absolutely no clue on what settings are required.

Is the above the only "flag" i need to add ? "CONFIG_USB_PULSE8_CEC=m"

I am working with the version 6.0 if cec and there are some flags one puts in - I am presuming I just ad this ? So delete everything, download it and ad this flag ?

Sorry for being vague - general new guy here.

heitbaum, 2020-11-09:
@audas  have a look at my changes to the kernel configuration (as contained in this commit) note: this is for the generic Pulse8 USB adapter.

https://github.com/LibreELEC/LibreELEC.tv/commit/47f0c272a4539b3516c6e93f4104382353edd0fb

audas, 2020-11-09:
Ok thanks - way above my skill set.   I will see if I can figure it out. Wish me luck.

closed 2025-02-27 by github-actions[bot]