Pulse-Eight/libcec #514: Windows fails to compile

ianmaddox, 2020-05-28:
Following the instructions on compiling for windows, I get the error "don't know how to make 'install'"
Running Windows 10
libcec version 5.0.0

Full log:
```
PS C:\libcec> .\windows\build-all.cmd
 * compiling libCEC libraries for amd64
 * compiling platform library for amd64
 --------------------------------------
 Generating cmake project:
 Architecture = amd64
 Project type = "NMake Makefiles"
 Cmake ARCH   =
 Project      = "C:\libcec\src\platform\windows\..\"
 Target       = "C:\libcec\windows\..\build\cmake\amd64"
 Install      = "C:\libcec\windows\..\build\amd64"
 Build type   = Release
 Toolchain    = Visual Studio 16 2019
 --------------------------------------

The system cannot find the path specified.
 Compiling "C:\libcec\windows\..\build\cmake\amd64" for amd64 using Visual Studio 16 2019

Microsoft (R) Program Maintenance Utility Version 14.25.28611.0
Copyright (C) Microsoft Corporation.  All rights reserved.

NMAKE : fatal error U1073: don't know how to make 'install'
Stop.
 * compiling libCEC for amd64
 --------------------------------------
 Generating cmake project:
 Architecture = amd64
 Project type = "NMake Makefiles"
 Cmake ARCH   =
 Project      = "C:\libcec\windows\.."
 Target       = "C:\libcec\windows\..\build\cmake\amd64"
 Install      = "C:\libcec\windows\..\build\amd64"
 Build type   = Release
 Toolchain    = Visual Studio 16 2019
 --------------------------------------

The system cannot find the path specified.
 *** failed to build libCEC for amd64 ***
Press any key to continue . . .
```

opdenkamp, 2020-07-10:
this was caused by one of the submodules that wasn't checked out when you built it. the latest version should now automatically pull all the submodules

closed 2020-07-10 by opdenkamp

rhyek, 2020-09-26:
I am getting the same error:
```
PS C:\Users\Carlos\Dev\third-party\pulse-eight> git clone git@github.com:Pulse-Eight/libcec.git
Cloning into 'libcec'...
remote: Enumerating objects: 219, done.
remote: Counting objects: 100% (219/219), done.
remote: Compressing objects: 100% (152/152), done.
remote: Total 15643 (delta 110), reused 138 (delta 67), pack-reused 15424
Receiving objects: 100% (15643/15643), 5.57 MiB | 12.65 MiB/s, done.
Resolving deltas: 100% (11820/11820), done.
PS C:\Users\Carlos\Dev\third-party\pulse-eight> cd .\libcec\
PS C:\Users\Carlos\Dev\third-party\pulse-eight\libcec> git submodule update --init --recursive
Submodule 'src/dotnet' (https://github.com/Pulse-Eight/cec-dotnet.git) registered for path 'src/dotnet'
Submodule 'src/platform' (https://github.com/Pulse-Eight/platform) registered for path 'src/platform'
Submodule 'support' (https://github.com/Pulse-Eight/libcec-support.git) registered for path 'support'
Cloning into 'C:/Users/Carlos/Dev/third-party/pulse-eight/libcec/src/dotnet'...
Cloning into 'C:/Users/Carlos/Dev/third-party/pulse-eight/libcec/src/platform'...
Cloning into 'C:/Users/Carlos/Dev/third-party/pulse-eight/libcec/support'...
Submodule path 'src/dotnet': checked out '30faf2b2b464d04ae8fc5892e90030404e56365a'
Submodule path 'src/platform': checked out '1133cccd412fd5dba541bf735f3facc5b72321c3'
Submodule 'support' (https://github.com/Pulse-Eight/libcec-support.git) registered for path 'src/platform/support'
Cloning into 'C:/Users/Carlos/Dev/third-party/pulse-eight/libcec/src/platform/support'...
Submodule path 'src/platform/support': checked out '416da1d0e60ee2397f31d4eb8f85886a9ef3327a'
Submodule path 'support': checked out '416da1d0e60ee2397f31d4eb8f85886a9ef3327a'
PS C:\Users\Carlos\Dev\third-party\pulse-eight\libcec> .\windows\build-all.cmd
 * compiling libCEC libraries for amd64
 * compiling platform library for amd64
 --------------------------------------
 Generating cmake project:
 Architecture = amd64
 Project type = "NMake Makefiles"
 Cmake ARCH   =
 Project      = "C:\Users\Carlos\Dev\third-party\pulse-eight\libcec\src\platform\windows\..\"
 Target       = "C:\Users\Carlos\Dev\third-party\pulse-eight\libcec\windows\..\build\cmake\amd64"
 Install      = "C:\Users\Carlos\Dev\third-party\pulse-eight\libcec\windows\..\build\amd64"
 Build type   = Release
 Toolchain    = Visual Studio 16 2019
 --------------------------------------

The system cannot find the path specified.
 Compiling "C:\Users\Carlos\Dev\third-party\pulse-eight\libcec\windows\..\build\cmake\amd64" for amd64 using Visual Studio 16 2019

Microsoft (R) Program Maintenance Utility Version 14.27.29111.0
Copyright (C) Microsoft Corporation.  All rights reserved.

NMAKE : fatal error U1073: don't know how to make 'install'
Stop.
 *** failed to build libCEC for amd64 ***
Press any key to continue . . .
```