Pulse-Eight/libcec #517: Remote Input Works Only Sometimes

wtf911, 2020-06-12:
Hello I have a Vizio TV with the USB CEC adapter. It works with the cec-tray app and with Kodi. Sometimes when I open one of them I can press the arrow keys on the remote to control the computer/Kodi and sometimes when I open them it doesn't work. I myself am not able to tell any difference of why sometimes it works and others it does not. Please let me know if there's something I do to help figure this out.

[image]

[cec-log.txt](https://github.com/Pulse-Eight/libcec/files/4772405/cec-log.txt)

wtf911, 2020-06-12:
Two things...

1. I should have mentioned I'm using firmware v12 and libcec 5.0.0

2. I figured out the pattern to when this happens. If the TV is already on and then I launch Kodi I can use the remote buttons to control Kodi. But if the TV is off and I launch Kodi, Kodi automatically turns on the TV, "libcec connected to TV" popup, then the buttons on my remote do not work to control Kodi. So for something in the programming/timing of things if Kodi opening causes the TV to power on then no luck with remote input working. But if the TV is on beforehand, I open Kodi, get the little "libcec connected to TV" then the buttons on the remote input to Kodi does work.

opdenkamp, 2020-07-10:
Kodi and cec-tray don't work well together, because only one of them can use the adapter.
Can you try with the latest version of libCEC: https://github.com/Pulse-Eight/libcec/releases/download/libcec-6.0.0/libcec-6.0.0.exe

closed 2025-02-27 by github-actions[bot]